MaJerle/ESP8266_AT_Commands_parser #2: Using your library with CubeMX

dinhdiep9947, 2016-03-10:
Dear Tilen, I programed for STM32F4 a month ago. I use to CubeMx to config STM32F4, and try to using your library. But do not it. So, do you have slove to me? I speak English very bad. Sorry about that.

MaJerle, 2016-03-10:
You dont need cubeMX to use this library. Just manually add files to project.

dinhdiep9947, 2016-03-10:
Thanks. But I'm newer, I use to CubeMX to config STM32F4 and then I want to program STM32F103. I write file user.c for declare I2C_HandleTypeDef hi2c1; In file main.c is geanered by CubeMX, when I build in Keil C, it have a error "L6200E". I know this error but i don't know fix this error. Do you give solve to me?
Thanks for read my question.

MaJerle, 2016-03-10:
Your problem has nothing to do with Issue message for this library (which is not and issue at all).

closed 2016-03-10 by MaJerle