MaJerle/ESP8266_AT_Commands_parser #4: Port to NuttX

drasko, 2016-04-12:
Hello @MaJerle,
I think it would be great to port this code to NuttX RTOS, so we can use it on wide variety of platforms. Can you please post a few instructions how to start and what is needed for NuttX RTOS port?

Can this code be published under BSD license (NuttX uses this license)?

BR,
Drasko

MaJerle, 2016-04-12:
Library is platform independent.
How to use it on your RTOS is up to you. You need to write low-level drivers for your MCU you are using.

Keep in mind, API functions are not thread-safe. You will need mutexes or semaphores to use this with RTOS.

drasko, 2016-04-12:
@MaJerle NuttX has already all HAL drivers, what is needed it to port your library to use NuttX API.

Functions must be re-written to be thread-safe.

What about the license? BSD?

MaJerle, 2016-04-12:
I dont know what api you needfrom there. You ha e to make sure about transmit and receive datato mcu. 3 functions. 

I dont know about license. You should check google or similar.

drasko, 2016-04-12:
No, I meant your code - under which license did you published it?

closed 2016-04-12 by MaJerle