MaJerle/ESP8266_AT_Commands_parser #6: +IPD Problem

omidtalebi, 2016-06-05:
Hi
ESP8266 respond "AT+......\r\n" for all command EXCEPT "+IPD" that it sends "\r\n+IPD,.....". Your code knows responses that have "\r\n" on their ends and discard alone "\r\n". So "\r\n" before the "+IPD" will be discarded and "+IPD" data will wait until next "\r\n+IPD,...." response that "\r\n" of next response is used as terminator of previous response.

MaJerle, 2016-06-05:
Hi, can you show data from esp when this happen? You can use pastebin or similar tool for copying code.

omidtalebi, 2016-06-06:
Please see the attached Image. I added some codes to function BUFFER_ReadString and solved this problem. It is attached in a docx file too.

[image]

[BUFFER_ReadString.docx](https://github.com/MaJerle/ESP8266_AT_Commands_parser/files/299798/BUFFER_ReadString.docx)

MaJerle, 2016-07-13:
This problem was easily to reproduce so I take a deep look at it and I think I fixed it.
I have commited fix to test it if you want. Instead of modifying default buffer options (which I don't want) I modified parsing procedure.

MaJerle, 2016-07-16:
Latest version fixes this issue.

closed 2016-07-16 by MaJerle