MaJerle/ESP8266_AT_Commands_parser #8: Is it possible to provide blocking data sending API?

Kevin0626, 2016-07-12:
It will be esier for programming the logic with blocking data sending API. 

To send a data in a callback makes my propram much more complicated.

MaJerle, 2016-07-12:
Having callback system is with reason here. The point is that this can take long time to respond and doing "nothing" that time (waiting response) is painful.

On the other hand, if you get response like 100kB long, you probably don't have so much memory to proceed everything at a time. 

2 main reasons why callbacks are used in small chunks of memory packets if RAM is a problem on your MCU.

Kevin0626, 2016-07-12:
recieving data in callback is perfect for my program. 

what I want is to support sending data in my main logic. 

Thanks..

MaJerle, 2016-07-12:
This gives you the same problem as mentioned first time, especially the one with RAM.
What you should do is this:
1. Start connection, give it unique name,
2. Request data send when connection is active (connected callback)
3. In callback to set data add data and send it
4. When callback is called for data sent, call again requestsenddata function.
5. Use state machine (simple counter) to detect what you have sent already and what you stil have.

Kevin0626, 2016-07-12:
we are porting LWM2M protocol to stm32 f103 board. The code maintains multiple timers and handle receieved data in a callbck like way. but it send data directly in many places. We can modify it to adapt the data sending to callback model, but the whole logic will be much harder to maintain.

closed 2016-07-28 by MaJerle