MaJerle/ESP8266_AT_Commands_parser #10: esp8266 stuck in waiting data

Kevin0626, 2016-07-13:
My code tries to initiate another data send in the callback ESP8266_Callback_ClientConnectionDataSent(). Then the esp8266 seems stuck in the waiting data status. It looks the parser missed the ">".

/* Called when data are send successfully */
void ESP8266_Callback_ClientConnectionDataSent(ESP8266_t* ESP8266, ESP8266_Connection_t* Connection) {
	printf("Data successfully sent as client!\r\n");
    
    ESP8266_RequestSendData(ESP8266, Connection);
}

======stm32 tx output ============

Wifi is disconnected!
Wifi is connected!
Wifi got an IP address
send: AT+CIPSTA_CUR?

Grabbing IP status: 0
We have valid IP address: 192.168.4.2
send: AT+CWJAP_CUR?

send: AT+CIPSTART=
send: 0
send: ,"UDP","
send: 192.168.4.4
send: ",
send: 8080
send: 

Client connected to server! Connection number: 192.168.4.4
send: AT+CIPSENDEX=
send: 0
send: ,2048

send: GET / HTTP/1.1
Host: stm32f4-discovery.com
Connection: close

send: \0
Data successfully sent as client!
send: AT+CIPSENDEX=
send: 0
send: ,2048

======  esp8266 tx output =====
OK
+CWJAP_CUR:"ESP_01154F","5e:cf:7f:01:15:4f",1,-15

OK
0,CONNECT

OK
17:11:28.966 
OK
> 
Recv 66 bytes

SEND OK

OK
>

MaJerle, 2016-07-13:
As I see from response, my stack did respond to your next RequestDataSend
call and it did send AT command to module.

Module did not respond with ">" character. This is something to investigate.
ᐧ

*Tilen Majerle*

   - Research and development engineer
   -

MESI, development of medical devices, LTD
Letališča cesta 3c
1000 Ljubljana
Slovenia

*P* 00386 1 620 34 87
*F* 00386 8 205 31 95
*W *mesimedical.com
*FB* facebook.com/MESIdoo

Kevin0626, 2016-07-13:
Majerle,

There is char "<" in my log actually. I didn't include it in my pasted log.

MaJerle, 2016-07-13:
Hm, I don't see a reason why would this happen from code.

Can you check what does ESP8266_RequestSendData function returns second time when stack fails?

Kevin0626, 2016-07-14:
I may find the root casue.  
firstly I ensure the ">" is recieved the by uart buffer. then I found it might be related with the value of ESP8266->Flags.F.WaitForWrapper.

I have printf all the places of change its value. and got log as below. It seem the "SEND OK' of last send was delayed for handle which caused the WaitForWrapper is reset to 0.

====== stm32 output  ======================

Client connected to server! Connection number: 192.168.4.4
send: AT+CIPSENDEX=
send: 0
send: ,2048

ESP8266_RequestSendData: WaitForWrapper=1
ESP8266_COMMAND_SEND: WaitForWrapper=1
ProcessSendData: WaitForWrapper=0
send: GET / HTTP/1.1
Host: stm32f4-discovery.com
Connection: close

send: \0
Data successfully sent as client!
send: AT+CIPSENDEX=
send: 0
send: ,2048

ESP8266_RequestSendData: WaitForWrapper=1
SEND OK: WaitForWrapper=0

====== esp 8266 tx output =====

OK
+CWJAP_CUR:"ESP_01154F","5e:cf:7f:01:15:4f",1,-33

OK
0,CONNECT

OK

OK
> 
Recv 66 bytes

SEND OK

OK
> 

Also I met another problem: in the ESP8266_Init(), after the stm32 send "at+rst" and the esp reset. The stm32 continue to run even esp tx has nothing monitored.

MaJerle, 2016-07-14:
As I see from your response, the problem is somewhere in `ESP8266_RequestSendData` function.
I think that second time, SendCommand function fails and returns you out and WaitingForWrapper is not set actually. Can you verify this for me if this is true?
Just try to print out something in `ESP8266_RequestSendData` function inside if statement for `SendCommand` function.

```
    if (SendCommand(ESP8266, ESP8266_COMMAND_SEND, NULL, NULL) != ESP_OK) {
        return ESP8266->Result;
    }
```

Kevin0626, 2016-07-14:
I can test it.

Don't you think the sequence of following two log lines is in problem?
ESP8266_RequestSendData: WaitForWrapper=1
SEND OK: WaitForWrapper=0

I called ESP8266_RequestSendData in ESP8266_Callback_ClientConnectionDataSent().

MaJerle, 2016-07-14:
If possible, add me via Skype (tjuln10) where we can check code in details.

It is possible the problem you said.

Kevin0626, 2016-07-14:
I added the test code in ESP8266_RequestSendData as you requested, and there is no trigger of it.

I also tried to commented out the flag clear for "SEND OK", the problem is still there.

	if (strcmp(Received, "SEND OK\r\n") == 0) {
		/* Force IDLE when we are in SEND mode and SEND OK is returned. Do not wait for "> " wrapper */
		ESP8266->ActiveCommand = ESP8266_COMMAND_IDLE;
        printf ("SEND OK: WaitForWrapper=0\n");
		
		/* Clear flag */
		//ESP8266->Flags.F.WaitForWrapper = 0;
	}

I don't have skype now. I will see if I can use it.

MaJerle, 2016-07-16:
Latest commit has fixed this issue.

closed 2016-07-16 by MaJerle