MaJerle/ESP8266_AT_Commands_parser #15: DelayMS

LowClue, 2016-09-07:
Hello, if i dont comment delay the program dont run after delay...

I have this warning, ¿What is the problem?

[image]

LowClue, 2016-09-07:
And when i use waitready dont wait an OK sometimes, i am wrong?

MaJerle, 2016-09-07:
Wait ready command waits stack to have active command IDLE. It does not matter if it is OK or ERROR response from ESP.

Regarding DELAYMS macro, this is something relevant with your compiler. Statements are ANSI C declared so I don't see a reason for warnings.

LowClue, 2016-09-07:
if i want that wait a ok,  for example, that is possible?

I want to do this because i want to do other functions and if i run my functions the ESP is busy when i want to do other order.

MaJerle, 2016-09-07:
You can wait a ready first and then check last status.
If ESP->Flags.F.LastOperationStatus then is OK else was error.

What other "order" you want to do that time for example?

LowClue, 2016-09-07:
**For example, i run this two function:

First:**

ESP8266_Result_t ESP8266_PostConfig(ESP8266_t* ESP8266) {
 
        char getcmd1[]="GET ....."
        getlenght=BTPS_StringLength(getcmd1);

        ESP8266_WaitReady(ESP8266);   /*This maybe is wrong, because i was think that wait to OK*/
   while (ESP8266_SetMode(ESP8266, ESP8266_Mode_STA_AP) != ESP_OK);   
   while (SetMux(ESP8266, 1) != ESP_OK);          

   ESP8266_WaitReady(ESP8266);
   while (ESP8266_WifiDisconnect(ESP8266) != ESP_OK);
   ESP8266_WaitReady(ESP8266);
   while(ESP8266_ListWifiStations(ESP8266) != ESP_OK);
   ESP8266_WaitReady(ESP8266);
   while(ESP8266_WifiConnect(ESP8266, "LockUp", "ff1479fFff1479fFes") != ESP_OK);
   ESP8266_WaitReady(ESP8266);
   while(ESP8266_WifiGetConnected(ESP8266)!= ESP_OK);
   ESP8266_WaitReady(ESP8266);
   ESP8266_SetSSLBufferSize(ESP8266,4096);
   ESP8266_WaitReady(ESP8266);
   ESP8266_StartClientConnectionSSL(ESP8266,"example","example",443,0);
   ESP8266_WaitReady(ESP8266);
   

    if (SendCommand(ESP8266, ESP8266_COMMAND_SEND, NULL, NULL) != ESP_OK) { /* Send command */
        return ESP8266->Result;
    } 
    
    return ESP8266->Result;                                 /* Return from function */
}

**This function its oky, because in start SSL go to callback void ESP8266_Callback_ClientConnectionConnected(ESP8266_t* ESP8266, ESP8266_Connection_t* Connection)  and i do that i want but in the second function:**

ESP8266_Result_t ESP8266_ConfigEsp8266(ESP8266_t* ESP8266) {
    ESP8266_CHECK_IDLE(ESP8266); 

        char ConfigWeb[]="<html><head></head><body>\r\n<h1>ESP8266 Configuacion</h1>\r\n<p>Servidor de PEPE</p>\r\n";
        getlenght=BTPS_StringLength(ConfigWeb);
    ESP8266_WaitReady(ESP8266);    
    while (SetMux(ESP8266, 1) != ESP_OK);
    ESP8266_ServerEnable(ESP8266, 80);
    ESP8266_WaitReady(ESP8266);
    while (Setdinfo(ESP8266, 1) != ESP_OK);              
    ESP8266_WaitReady(ESP8266);
    ESP8266_GetConnectedStations(ESP8266);
    ESP8266_WaitReady(ESP8266);

**// In this part, i do manually Callback for +IPD, because i have problems to detect when in web browser i put the AP and in console i can see my +IPD is not detect, then do that manually.... //**

    ESP8266_Callback_ServerConnectionDataReceived(ESP8266, &ESP8266->Connection[ESP8266->IPD.ConnNumber], ESP8266->Connection[ESP8266->IPD.ConnNumber].Data);
    return ESP8266->Result;                              
}

**I have problems with the time if i want to do CIPSENDS and dont wait to 'SEND OK' for do other cipsend for example.**

A lot of thank another time.

MaJerle, 2016-09-07:
Must say I don't understand.

LowClue, 2016-09-07:
**For example, i run this two function:**

**First, i want to do one get in my server, then configure ESP to conection with server with SSLconnection:**

```
ESP8266_Result_t ESP8266_PostConfig(ESP8266_t* ESP8266) {
 
        char getcmd1[]="GET ....."
        getlenght=BTPS_StringLength(getcmd1);

        ESP8266_WaitReady(ESP8266);   /*This maybe is wrong, because i was think that wait to OK*/
   while (ESP8266_SetMode(ESP8266, ESP8266_Mode_STA_AP) != ESP_OK);   
   while (SetMux(ESP8266, 1) != ESP_OK);          

   ESP8266_WaitReady(ESP8266);
   while (ESP8266_WifiDisconnect(ESP8266) != ESP_OK);
   ESP8266_WaitReady(ESP8266);
   while(ESP8266_ListWifiStations(ESP8266) != ESP_OK);
   ESP8266_WaitReady(ESP8266);
   while(ESP8266_WifiConnect(ESP8266, "LockUp", "ff1479fFff1479fFes") != ESP_OK);
   ESP8266_WaitReady(ESP8266);
   while(ESP8266_WifiGetConnected(ESP8266)!= ESP_OK);
   ESP8266_WaitReady(ESP8266);
   ESP8266_SetSSLBufferSize(ESP8266,4096);
   ESP8266_WaitReady(ESP8266);
   ESP8266_StartClientConnectionSSL(ESP8266,"example","example",443,0);
   ESP8266_WaitReady(ESP8266);
   

    if (SendCommand(ESP8266, ESP8266_COMMAND_SEND, NULL, NULL) != ESP_OK) { /* Send command */
        return ESP8266->Result;
    } 
    
    return ESP8266->Result;                                 /* Return from function */
}

ESP8266_Callback_ClientConnectionConnected(ESP8266_t* ESP8266, ESP8266_Connection_t* Connection) {
ESP8266_RequestSendData(ESP8266,Connection);
} 
```

**This function its oky, because in start SSL go to callback void, and go to requestsenddata that do a cipsend, this oky. In the second configuration i want to do an acces point: **

```
ESP8266_Result_t ESP8266_ConfigEsp8266(ESP8266_t* ESP8266) {
    ESP8266_CHECK_IDLE(ESP8266); 

        char ConfigWeb[]="<html><head></head><body>\r\n<h1>ESP8266 Configuacion</h1>\r\n<p>Servidor de PEPE</p>\r\n";
        getlenght=BTPS_StringLength(ConfigWeb);
    ESP8266_WaitReady(ESP8266);    
    while (SetMux(ESP8266, 1) != ESP_OK);
    while (Setdinfo(ESP8266, 1) != ESP_OK);              
    ESP8266_WaitReady(ESP8266);
    ESP8266_GetConnectedStations(ESP8266);
    ESP8266_WaitReady(ESP8266);

```

**In this part i do breakpoint, because i need to do a linked with accespoint to do a cipsend, when the code stop in breakpoint i go to web browser to do a +IPD and link (code should to detect +IPD autamatic, but dont happend for that the breakpoint and call callback manually).**

```
    ESP8266_Callback_ServerConnectionDataReceived(ESP8266, &ESP8266->Connection[ESP8266->IPD.ConnNumber], ESP8266->Connection[ESP8266->IPD.ConnNumber].Data);
    return ESP8266->Result;                              
}

```
[image]

**A lot of thank another time.**

LowClue, 2016-09-08:
In conclusion Majerle and most importantly, it does not detect the callback send ok, + IPD or x, closed for example.

I need these callbacks to work with this library. In the last image, you can see one +IPD and code does not detect that...

Thanks, sorry for my english.

MaJerle, 2016-09-08:
As far I see, you did library modifications so I'm not responsible for your problems anymore.
Original lib works as supposed in general case.

closed 2016-09-15 by MaJerle