MaJerle/ESP8266_AT_Commands_parser #16: ESP8266_DataReceived is not interrupt safe

jimparis, 2016-09-13:
In `01-ESP8266_STM32Fxxx/User/esp8266_ll.c`, you have:
```
/* USART receive interrupt handler */
void TM_USART1_ReceiveHandler(uint8_t ch) {
	/* Send received character to ESP stack */
	ESP8266_DataReceived(&ch, 1);
}
```
But `ESP8266_DataReceived` is not interrupt safe because `BUFFER_Write` is not re-entrant.  For example, if a character is received while `ESP8266_Update` is in the middle of its own `BUFFER_Write` call, the buffer will be corrupted.

MaJerle, 2016-09-13:
It is not reentrant call to buffer write, but normal buffer write is executed only from one (1) location and this is data received so it is interrupt save.

jimparis, 2016-09-13:
No, BUFFER_Write is called from more than one location.  See ESP8266_Update for example.

jimparis, 2016-09-13:
Actually it's BUFFER_WriteToTop.  But that has the same problem.  For example:

- `BUFFER_WriteToTop` calls `BUFFER_GetFree`. 
- Interrupt happens, `ESP8266_DataReceived` is called, and the buffer entirely fills up with new data
- Now `BUFFER_WriteToTop` proceeds to overwrite the `count` bytes of the just-received data!

MaJerle, 2016-09-14:
Free and full functions are reentrant. 

Also, there is a define to specify buffer size for tmp and non tmp buffers. Its app specific to setup these buffers but. If you find this problem in real situation, let me know.

jimparis, 2016-09-14:
It doesn't matter if free and full functions are reentrant.  In my example above, I described a situation where the interrupt occurs immediately after GetFree returns, not in the middle of it.

This also is unrelated to temp buffers.  The example I described is operating solely on USART_Buffer.

I'm definitely seeing this problem in real situations.  But it is hard to give specifics because race conditions are difficult to pin down, especially when two asynchronous CPUs are involved.  I gave an example problem above but I can try to come up with more.  Would a demonstration on Linux be helpful?

MaJerle, 2016-09-14:
WriteToTop writes on top of buffer while DataReceived writes to bottom of it. So this can happen when buffer is too small and you can fix that by increasing it.

This lib is not yet done for multitasking so it is not relevant right now.

jimparis, 2016-09-14:
Calling DataReceived from an interrupt *is* multitasking.  It can happen at any time.  That is the problem I'm seeing.

I will come up with a better example when I get into work today.  I think I can show the same problem with read or find getting interrupted.

MaJerle, 2016-09-14:
Ok, I give up here. The point is that every operation is done from 1 (SINGLE) location. So reading from one location, write from one location and write to top from one location. So I'm always writing to bottom of buffer and if required, I write to top of buffer, again only from one location.

If your processor is too slow to handle all that, you have to make buffer larger.
And this works ok. Let me know about real situation when this won't work with buffer enough in size.

But again, I already said I will rewrite Update function to handle bytes other way but looks like you ignore that statement before.

jimparis, 2016-09-14:
I didn't ignore it, but I am not sure that it would help.  Bear with me a bit and I will try to come up with a more clear example.  I may be wrong but I'm definitely seeing problems in real situations, with a fast processor and big buffers.  I will dig into this over the next few hours and let you know what I find.  Thanks.

jimparis, 2016-09-14:
You are right.  After 4 more hours of pulling my hair out I discovered that this was actually a race in mbed's `Stream` implementation where I was getting my serial data.  And the rewrite you mentioned should fix the race I described [above](https://github.com/MaJerle/ESP8266_AT_Commands_parser/issues/16#issuecomment-246850351).  Thanks for your fast response.

closed 2016-09-14 by jimparis