MaJerle/ESP8266_AT_Commands_parser #17: Why not use "AT+CIPSENDBUF" to improve SEND performance?

bxwllzz, 2016-09-21:
"AT+CIPSENDEX" is blocking and slow. I used to write a esp8266 AT cmd parser using "AT+CIPSENDBUF" and implement it in STM32F103 and Kinetis K60. I set UART baudrate to 4.5Mbps and get TCP upload speed of 220kB/s+(from STM32 with ESP8266 to PC), TCP download speed of more than 300kB/s.

MaJerle, 2016-09-21:
Maybe you are rught regarding sendbuf or sendsex commands. But you should know some things here:

1. Download speed has nothing to do with CIPSENDEX or CIPSENDBUF,  your actual internet connection and baudrate has sense here. With 921600bauds I get little less than 100k (around 96k) Bytes, which is pretty much highestpossible with that baudrate. 

2. Library is designed as non-blocking so it actually doesnt care if command is slow. There are events to handle all that for you. 

3. Handling CIPSENDBUF is more complicated than without that. 

But anyway, your concept can be better than mine but it still is not so much because of facts I gave you above. 

Im still ready to discuss regarding this topic.

bxwllzz, 2016-09-22:
I write a simple python script to test the upload speed of "AT+CIPSEND", "AT+CIPSENDEX" and "AT+CIPSENDBUF". 
PROGRAM:
`import serial
import time

def wait_custom(str, display=True):
    start_time = time.time()
    dat = b''
    while dat[-len(str):] != str:
        new_dat = ser.read(1)
        if display:
            print(new_dat.decode('utf8'), end='')
        elif time.time() - start_time > 10:
            print('timeout')
            print(dat.decode('utf8'))
            input()
        dat += new_dat

def wait_OK_ERROR(display=True):
    start_time = time.time()
    dat = b''
    while dat[-6:] != b'\r\nOK\r\n' and dat[-9:] != b'\r\nERROR\r\n':
        new_dat = ser.read(1)
        if display:
            print(new_dat.decode('utf8'), end='')
        elif time.time() - start_time > 10:
            print('timeout')
            print(dat.decode('utf8'))
            input()

        dat += new_dat
    if dat[-6:] == b'\r\nOK\r\n':
        return True
    else:
        return False

def AT_command(cmd):
    ser.read(1024)
    print(cmd.decode('utf8'))
    ser.write(cmd + b'\r\n')
    wait_OK_ERROR()
    
def AT_send(dat):
    cmd = 'AT+CIPSEND=0,%d'%len(dat)
    ser.write(cmd.encode('utf8') + b'\r\n')
    if wait_OK_ERROR(False):
        wait_custom(b'> ', False)
        ser.write(dat)
        wait_custom(b'\r\nSEND OK\r\n', False)
        return True
    else:
        return False
    
def AT_sendex(dat):
    cmd = 'AT+CIPSENDEX=0,2048'
    ser.write(cmd.encode('utf8') + b'\r\n')
    if wait_OK_ERROR(False):
        wait_custom(b'> ', False)
        ser.write(dat)
        ser.write(b'\\0')
        wait_custom(b'\r\nSEND OK\r\n', False)
        return True
    else:
        return False
 
def AT_sendbuf(dat):
    cmd = 'AT+CIPSENDBUF=0,%d'%len(dat)
    ser.write(cmd.encode('utf8') + b'\r\n')
    if wait_OK_ERROR(False):
        wait_custom(b'> ', False)
        ser.write(dat)
        wait_custom(b'bytes\r\n', False)
        return True
    else:
        return False
    
    
if __name__ == '__main__':
    ser = serial.Serial(10, baudrate=115200, timeout=0)
    AT_command(b'ATE0')
    AT_command(b"AT+GMR")
    # set baudrate to 2Mbps
    AT_command(b'AT+UART_CUR=2000000,8,1,0,0')
    ser.close()
    ser = serial.Serial(10, baudrate=2000000, timeout=0)
    # config wifi
    AT_command(b'AT+CWMODE_DEF=3')
    AT_command(b'AT+CWJAP_DEF="HC5761","shuixiang"')
    AT_command(b'AT+CWSAP_DEF="ESP8266","12345678",7,0,4,0')
    AT_command(b'AT+CWAUTOCONN=1')
    # config tcp server
    AT_command(b"AT+CIPMUX=1")
    AT_command(b'AT+CIPSERVER=1,23')
    AT_command(b'AT+CIPSTO=5')
    # config tcp client to my PC
    AT_command(b'AT+CIPSTART=0,"TCP","192.168.123.186",23,0')

    frame_size = 2000
    frame_count = 100

    # AT+CIPSEND speedtest
    print('AT+CIPSEND speedtest')
    start_time = time.time()
    for i in range(frame_count):
        while AT_send(b'-'*frame_size) == False:
            pass
    duration = time.time() - start_time
    speed = frame_size * frame_count / duration / 1000
    print("total sent %d * %d = %d bytes"%(frame_size, frame_count, frame_size * frame_count))
    print('use %.3f sec'%duration)
    print('speed: %.3fkB/s'%speed)
    
    # AT+CIPSENDEX speedtest
    print('AT+CIPSENDEX speedtest')
    start_time = time.time()
    for i in range(frame_count):
        while AT_sendex(b'-'*frame_size) == False:
            pass
    duration = time.time() - start_time
    speed = frame_size * frame_count / duration / 1000
    print("total sent %d * %d = %d bytes"%(frame_size, frame_count, frame_size * frame_count))
    print('use %.3f sec'%duration)
    print('speed: %.3fkB/s'%speed)

    # AT+CIPSENDBUF speedtest
    print('AT+CIPSENDBUF speedtest')
    start_time = time.time()
    for i in range(frame_count):
        while AT_sendbuf(b'-'*frame_size) == False:
            pass
    duration = time.time() - start_time
    speed = frame_size * frame_count / duration / 1000
    print("total sent %d * %d = %d bytes"%(frame_size, frame_count, frame_size * frame_count))
    print('use %.3f sec'%duration)
    print('speed: %.3fkB/s'%speed)
    
    ser.close()
    
`
OUTPUT:
ATE0
ATE0

OK

AT+GMR
AT version:1.3.0.0(Jul 14 2016 18:54:01)

SDK version:2.0.0(656edbf)

compile time:Jul 19 2016 18:44:44

OK

AT+UART_CUR=2000000,8,1,0,0

OK

AT+CWMODE_DEF=3

OK

AT+CWJAP_DEF="HC5761","shuixiang"
WIFI CONNECTED

WIFI GOT IP

OK

AT+CWSAP_DEF="ESP8266","12345678",7,0,4,0

OK

AT+CWAUTOCONN=1

OK

AT+CIPMUX=1

OK

AT+CIPSERVER=1,23

OK

AT+CIPSTO=5

OK

AT+CIPSTART=0,"TCP","192.168.123.186",23,0
0,CONNECT

OK

AT+CIPSEND speedtest
total sent 2000 * 100 = 200000 bytes
use 4.337 sec
speed: 46.112kB/s
AT+CIPSENDEX speedtest
total sent 2000 * 100 = 200000 bytes
use 4.159 sec
speed: 48.084kB/s
AT+CIPSENDBUF speedtest
total sent 2000 * 100 = 200000 bytes
use 2.855 sec
speed: 70.063kB/s

MaJerle, 2016-09-23:
I would also like to see responses and times when did that happen for buffers sent.

bxwllzz, 2016-09-24:
Test program rewrite in C. 
Test data: 1400 bytes * 100 frame

AT+SEND OUTPUT (about 53kB/s): 
```
================= AT+SEND Test =================
[  0.083ms]AT+CIPSEND=0,1400
[  2.002ms]
[  2.036ms]OK
[  2.049ms]> [ 10.962ms][data][ 12.950ms]
[ 13.004ms]Recv 1400 bytes
[ 30.031ms]
[ 30.104ms]SEND OK
[ 30.207ms]AT+CIPSEND=0,1400
[ 32.063ms]
[ 32.096ms]OK
[ 32.107ms]> [ 41.028ms][data][ 42.983ms]
[ 43.029ms]Recv 1400 bytes
[ 56.631ms]
[ 56.676ms]SEND OK
[ 56.755ms]AT+CIPSEND=0,1400
[ 58.313ms]
[ 58.348ms]OK
[ 58.361ms]> [ 67.301ms][data][ 69.248ms]
[ 69.298ms]Recv 1400 bytes
[ 85.991ms]
[ 86.035ms]SEND OK
[ 86.112ms]AT+CIPSEND=0,1400
[ 87.709ms]
[ 87.745ms]OK
[ 87.755ms]> [ 96.748ms][data][ 98.604ms]
[ 98.652ms]Recv 1400 bytes
[102.700ms]
[102.734ms]SEND OK
[102.807ms]AT+CIPSEND=0,1400
[104.389ms]
[104.414ms]OK
[104.426ms]> [113.390ms][data][115.363ms]
[115.412ms]Recv 1400 bytes
[127.611ms]
[127.654ms]SEND OK
[127.728ms]AT+CIPSEND=0,1400
[129.316ms]
[129.356ms]OK
[129.368ms]> [138.307ms][data][140.248ms]
[140.305ms]Recv 1400 bytes
[153.208ms]
[153.248ms]SEND OK
[153.317ms]AT+CIPSEND=0,1400
[154.918ms]
[154.948ms]OK
[154.957ms]> [163.878ms][data][165.989ms]
[166.041ms]Recv 1400 bytes
[180.914ms]
[180.957ms]SEND OK
[181.032ms]AT+CIPSEND=0,1400
[182.575ms]
[182.606ms]OK
[182.617ms]> [191.547ms][data][193.507ms]
[193.557ms]Recv 1400 bytes
...more...
```

AT+SENDEX OUTPUT (about 53kB/s): 
```
================= AT+SENDEX Test =================
[  0.057ms]AT+CIPSENDEX=0,2048
[  1.595ms]
[  1.612ms]OK
[  1.619ms]> [ 10.592ms][data]\0[ 12.579ms]
[ 12.609ms]Recv 1400 bytes
[ 28.579ms]
[ 28.614ms]SEND OK
[ 28.672ms]AT+CIPSENDEX=0,2048
[ 30.274ms]
[ 30.290ms]OK
[ 30.297ms]> [ 39.245ms][data]\0[ 41.207ms]
[ 41.240ms]Recv 1400 bytes
[ 56.227ms]
[ 56.273ms]SEND OK
[ 56.344ms]AT+CIPSENDEX=0,2048
[ 57.925ms]
[ 57.949ms]OK
[ 57.960ms]> [ 66.908ms][data]\0[ 68.982ms]
[ 69.014ms]Recv 1400 bytes
[ 80.810ms]
[ 80.847ms]SEND OK
[ 80.920ms]AT+CIPSENDEX=0,2048
[ 82.503ms]
[ 82.530ms]OK
[ 82.540ms]> [ 91.439ms][data]\0[ 93.484ms]
[ 93.580ms]Recv 1400 bytes
[106.745ms]
[106.780ms]SEND OK
[106.862ms]AT+CIPSENDEX=0,2048
[108.427ms]
[108.446ms]OK
[108.456ms]> [117.378ms][data]\0[119.487ms]
[119.520ms]Recv 1400 bytes
[131.325ms]
[131.362ms]SEND OK
[131.420ms]AT+CIPSENDEX=0,2048
[133.039ms]
[133.075ms]OK
[133.087ms]> [142.004ms][data]\0[143.993ms]
[144.034ms]Recv 1400 bytes
[162.041ms]
[162.085ms]SEND OK
[162.160ms]AT+CIPSENDEX=0,2048
[163.737ms]
[163.776ms]OK
[163.788ms]> [172.712ms][data]\0[174.673ms]
[174.719ms]Recv 1400 bytes
[193.112ms]
[193.155ms]SEND OK
[193.226ms]AT+CIPSENDEX=0,2048
[194.832ms]
[194.881ms]OK
[194.891ms]> [203.790ms][data]\0[205.734ms]
...more...
```

AT+SENDBUF OUTPUT (less than 104kB/s): 
```
[  0.085ms]AT+CIPSENDBUF=0,1400
[  1.908ms]51,50
[  1.921ms]
[  1.930ms]OK
[  1.938ms]> [ 10.887ms][data][ 12.989ms]
[ 13.016ms]Recv 1400 bytes
[ 13.112ms]AT+CIPSENDBUF=0,1400
[ 14.883ms]52,50
[ 14.895ms]
[ 14.904ms]OK
[ 14.912ms]> [ 23.801ms][data][ 25.809ms]
[ 25.834ms]Recv 1400 bytes
[ 25.896ms]AT+CIPSENDBUF=0,1400
[ 27.858ms]
[ 27.885ms]0,51,SEND OK
[ 27.897ms]53,51
[ 27.903ms]
[ 27.911ms]OK
[ 27.917ms]> [ 36.830ms][data][ 38.252ms]
[ 38.278ms]Recv 1400 bytes
[ 38.340ms]AT+CIPSENDBUF=0,1400
[ 38.357ms]
[ 40.212ms]0,52,SEND OK
[ 40.231ms]54,52
[ 40.238ms]
[ 40.246ms]OK
[ 40.252ms]> [ 49.188ms][data][ 51.123ms]
[ 51.149ms]Recv 1400 bytes
[ 51.244ms]AT+CIPSENDBUF=0,1400
[ 52.137ms]BUFFER FULL
[ 52.149ms]
[ 52.161ms]ERROR
[ 52.268ms]AT+CIPSENDBUF=0,1400
[ 52.288ms]
[ 54.168ms]0,53,SEND OK
[ 54.185ms]55,53
[ 54.196ms]
[ 54.206ms]OK
[ 54.214ms]> [ 63.197ms][data][ 64.728ms]
[ 64.759ms]Recv 1400 bytes
[ 64.854ms]AT+CIPSENDBUF=0,1400
[ 64.871ms]
[ 66.987ms]0,54,SEND OK
[ 67.012ms]56,54
[ 67.020ms]
[ 67.031ms]OK
[ 67.039ms]> [ 75.985ms][data][ 77.760ms]
[ 77.796ms]Recv 1400 bytes
[ 77.928ms]AT+CIPSENDBUF=0,1400
[ 80.177ms]
[ 80.199ms]0,55,SEND OK
[ 80.214ms]57,55
[ 80.223ms]
[ 80.232ms]OK
[ 80.240ms]> [ 89.157ms][data][ 90.738ms]
[ 90.766ms]Recv 1400 bytes
[ 90.846ms]AT+CIPSENDBUF=0,1400
[ 90.861ms]
[ 92.731ms]0,56,SEND OK
[ 92.750ms]58,56
[ 92.759ms]
[ 92.768ms]OK
[ 92.777ms]> [101.700ms][data][103.612ms]
[103.651ms]Recv 1400 bytes
[103.717ms]AT+CIPSENDBUF=0,1400
[105.338ms]BUFFER FULL
[105.349ms]
[105.360ms]ERROR
[105.434ms]AT+CIPSENDBUF=0,1400
[107.069ms]BUFFER FULL
[107.079ms]
[107.090ms]ERROR
[107.160ms]AT+CIPSENDBUF=0,1400
[108.762ms]BUFFER FULL
[108.778ms]
[108.791ms]ERROR
[108.896ms]AT+CIPSENDBUF=0,1400
[111.182ms]
[111.211ms]0,57,SEND OK
[111.224ms]59,57
[111.234ms]
[111.244ms]OK
[111.251ms]> [120.172ms][data][121.254ms]
[121.298ms]Recv 1400 bytes
[121.378ms]AT+CIPSENDBUF=0,1400
[121.397ms]
[123.110ms]0,58,SEND OK
[123.147ms]60,58
[123.161ms]
[123.174ms]OK
[123.182ms]> [132.163ms][data][133.319ms]
[133.356ms]Recv 1400 bytes
[133.443ms]AT+CIPSENDBUF=0,1400
[133.455ms]
[135.408ms]0,59,SEND OK
[135.431ms]61,59
[135.439ms]
[135.450ms]OK
[135.458ms]> [144.413ms][data][145.086ms]
[145.113ms]0,60,SEND OK
[145.123ms]
[146.433ms]Recv 1400 bytes
[146.555ms]AT+CIPSENDBUF=0,1400
[148.356ms]62,60
[148.374ms]
[148.385ms]OK
[148.393ms]> [157.348ms][data][157.981ms]
[158.003ms]0,61,SEND OK
[158.011ms]
[159.367ms]Recv 1400 bytes
[159.448ms]AT+CIPSENDBUF=0,1400
[161.333ms]63,61
[161.349ms]
[161.359ms]OK
[161.367ms]> [170.371ms][data][170.974ms]
[170.998ms]0,62,SEND OK
[171.007ms]
[172.333ms]Recv 1400 bytes
[172.443ms]AT+CIPSENDBUF=0,1400
[174.332ms]64,62
[174.348ms]
[174.358ms]OK
[174.366ms]> [183.368ms][data][183.973ms]
[183.996ms]0,63,SEND OK
[184.006ms]
[185.202ms]Recv 1400 bytes
[185.304ms]AT+CIPSENDBUF=0,1400
[186.902ms]65,63
[186.915ms]
[186.924ms]OK
[186.930ms]> [195.845ms][data][197.840ms]
[197.883ms]Recv 1400 bytes
[197.949ms]AT+CIPSENDBUF=0,1400
[198.937ms]BUFFER FULL
[198.955ms]
[198.969ms]ERROR
[199.062ms]AT+CIPSENDBUF=0,1400
[199.080ms]
...more...
```
Full Output File: 
[output.txt](https://github.com/MaJerle/ESP8266_AT_Commands_parser/files/491012/output.txt)

MaJerle, 2016-11-10:
CIPSENDBUF is not used (not right now) because of hard to use parsing processing. That's the main reason. For good processing, you need high amount of RAM because of nature of CIPSENDBUF command.