MaJerle/ESP8266_AT_Commands_parser #26: ESP8266_SetAP

meros, 2016-10-17:
I am trying to use SetAP, looking at https://cdn.sparkfun.com/assets/learn_tutorials/4/0/3/4A-ESP8266__AT_Instruction_Set__EN_v0.30.pdf and trying it with my module I don't understand why the library sends maxconnections and hidden in this command. It gives error for my module, are there other versions of the at firmware that accepts these parameters or is this an error?

MaJerle, 2016-10-17:
[image]

I would say it works as it is expected.

meros, 2016-10-17:
I see, this is 0.51. I Am still on 0.40. Beside that I should update my firmware, do you have an oppinion on what to support? Only the latest? I guess trying to support all old versions is a lot of extra work...

MaJerle, 2016-10-17:
This is 1.5.1 version. I really don't see a need to support back versions. Things should be up to date as every new version fixes bugs in their software right now.

closed 2016-11-08 by MaJerle