MaJerle/ESP8266_AT_Commands_parser #27: CRLF after +IPD

nsalhuana, 2016-10-27:
Has there been any progress on this issue?

MaJerle, 2016-10-27:
Yes. This should work now.

nsalhuana, 2016-10-28:
I have still not been able to get it to receive the data. I have been using the SSL connection to client. I tested with both Google.com (works) and Firebase (doesn't work). I am using an FTDI board to see what the ESP8266 is trying to transmit.

Firebase
Console:
`Client connected to server! Connection number: thermo-devices

Data successfully sent as client!`

ESP8266:
`AT+CIPSENDEX=0,2048

OK
> 
Recv 86 bytes

SEND OK

+IPD,0,214,104.154.97.9,443:HTTP/1.1 200 OK
Server: nginx
Date: Fri, 28 Oct 2016 03:19:46 GMT
Content-Type: application/json; charset=utf-8
Content-Length: 27
Access-Control-Allow-Origin: *
Cache-Control: no-cache
Connection: close

+IPD,0,27,104.154.97.9,443:{"nics_dev":{"curTemp":69}}0,CLOSED`

Google.com
Console:
`Client connected to server! Connection number: google.com

Data successfully sent as client!
Data received from server on connection: google.com; Number of bytes received: 1387; 1387 / 0;
This is first packet received. Content length on this connection is: 0
Data received from server on connection: google.com; Number of bytes received: 1387; 2774 / 0;
Data received from server on connection: google.com; Number of bytes received: 1387; 4161 / 0;
Data received from server on connection: google.com; Number of bytes received: 1387; 5548 / 0;
Data received from server on connection: google.com; Number of bytes received: 1387; 6935 / 0;
Data received from server on connection: google.com; Number of bytes received: 1387; 8322 / 0;
Data received from server on connection: google.com; Number of bytes received: 1387; 9709 / 0;`

ESP8266:
`AT+CIPSENDEX=0,2048

OK
> 
Recv 59 bytes

SEND OK

+IPD,0,1387,216.58.209.4,443:HTTP/1.1 200 OK
Date: Fri, 28 Oct 2016 03:23:33 GMT
Expires: -1
Cache-Control: private, max-age=0
Content-Type: text/html; charset=ISO-8859-1
P3P: CP="This is not a P3P policy! See https://www.google.com/support/accounts/answer/151657?hl=en for more info."
Server: gws
X-XSS-Protection: 1; mode=block
X-Frame-Options: SAMEORIGIN
Set-Cookie: NID=89=QEZInQZQlR1BhqUIcpohyL9YLW5XNH-e8Wm3YuTVVoW1COxbVuALAJyfYoRKN47hjqG7sW8809isHqknx5jtYbo6tpp8c7FcmFRxCVgc3WtAlwYy30nlV99Sk2jCmi7pWRsRd4Us1CFV0A; expires=Sat, 29-Apr-2017 03:23:33 GMT; path=/; domain=.google.com; HttpOnly
Alt-Svc: quic=":443"; ma=2592000; v="36,35,34"
Accept-Ranges: none
Vary: Accept-Encoding
Connection: close

<!doctype html><html itemscope="" itemtype="http://schema.org/WebPage" lang="en"><head><meta content="Search the world's information, including webpages, images, videos and more. Google has many special features to help you find exactly what you're looking for." name="description"><meta content="noodp" name="robots"><meta content="text/html; charset=UTF-8" http-equiv="Content-Type"><meta content="/images/branding/googleg/1x/googleg_standard_color_128dp.png" itemprop="image"><title>Google</title><script>(function(){window.google={kEI:'tcQSWLyLFIOQUdStnvAK',kEXPI:'1351459,1351827,1351903,1351980,3700247,3700400,4029815,4031109,4032678,4036527,4038012,4039268,4043492,4045841,4048347,40`

MaJerle, 2016-10-28:
I see, still a problem here.
Will test it again and see the problem but this should work since I already made an update just for that.

Do you use latest version?

Anyway, I'm not satisfied with this lib, I will have to rewrite it. Will use technique I use in GSM library https://github.com/MaJerle/GSM_AT_commands_parser

nsalhuana, 2016-10-30:
Yes, I am using the latest version of the repo. I am also using the following:

AT FIrmware: 1.3.0
SDK Version: 2.0.0
Board: STM32F469I-Discovery

MaJerle, 2016-10-30:
This is strange, very strange.

Btw, today I started with new lib version 2.0 with improved parsing system. I will release alpha prerelease hopefully this week. It will have the same system as GSM AT parser.

nsalhuana, 2016-10-31:
Are you seeing this issue on your setup?

Great. I am looking forward to it and can help test it! Thanks

MaJerle, 2016-10-31:
I saw this issue while ago but I fixed that and was working ok. But I'm sure there are some corners where it might not work.

MaJerle, 2016-11-08:
New version has fixed this problems completelly because parser works other way.

closed 2016-11-08 by MaJerle