MaJerle/ESP8266_AT_Commands_parser #28: strlen of NULL pointer is generating a crash

marcelobarrosalmeida, 2016-10-30:
Hi ! 

I am creating a small workaround for running the AT parser at Linux, testing my code before putting it in a micro controller (ESP8266 is connect via USB/Serial adapter). 

During these testes, I found a problem related to strlen argument.  In esp8266.c, line 942, we have a call to strlen9ESP8266->ActiveCommandResponse). As ActiveCommandResponse is NULL since the current command is CIPMUX, a core dump is generated (gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.2) )

```
// ...
 strncmp(Received, ESP8266->ActiveCommandResponse, strlen(ESP8266->ActiveCommandResponse) ) != 0
// ...
```

I did the following (ugly) workaround:

```
// ...
strncmp(Received, ESP8266->ActiveCommandResponse, ESP8266->ActiveCommandResponse ? strlen(ESP8266->ActiveCommandResponse) : 0) != 0
// ...
```

Maybe a macro could be fine and better solution. Please, take a look. The following reference may help:
http://stackoverflow.com/questions/5796103/strlen-not-checking-for-null

marcelobarrosalmeida, 2016-10-31:
Standart used was gnu99 instead c99 due to a gnu extensions used in the code.

MaJerle, 2016-11-08:
This is actually a very nice catch. I have rewritten ESP code and this feature is not usable anymore.

closed 2016-11-08 by MaJerle