MaJerle/ESP8266_AT_Commands_parser #29: ESP_Init - Device does not reset in While Loop

nsalhuana, 2016-11-08:
Hi Tilen,

I had the previous version of this library working and have begun testing the new version. The program gets stuck waiting in this while loop in "esp8266.c":
```
    while (i--) {
    	printf("In the loop\n");
        __ACTIVE_CMD(ESP, CMD_BASIC_RST);                   /* Reset device */
        ESP_WaitReady(ESP, ESP->ActiveCmdTimeout);
        __IDLE(ESP);
        if (ESP->ActiveResult == espOK) {
            break;
        }
    }
```

I have it printing while it is in the loop but it never gets past this. I have changed "esp8266_ll.c" to use USART6 with PinPack-2. After the Reset times out in the while loop, I see that I am sending "AT" and receiving "OK" which shows that the USART6 is set up correctly. Please advise.

Thank you for all your hard work.

MaJerle, 2016-11-08:
If you see AT\r\n sent then your low-level driver works ok.
How is receive part done? Did you do UART interrupt handler?

When using my examples for STM32, I use UART1 for communication. In esp8266_ll.c file for example is interrupt handler TM_USART1_ReceiveHandler and is called if it is enabled. 
Check defines.h file in example (any example on this page) what is there.

I think this will solve a problem. If you don't know what to do, ask agian.

nsalhuana, 2016-11-08:
Thank you for the quick response. I did a search for all the locations that you used USART1 and replaced them with USART6 for receiving.

My esp8266_ll.c is as follows (disregard the debug printfs):
```
uint8_t ESP_LL_Init(ESP_LL_t* LL) {
    /* Init USART */
    TM_USART_Init(USART6, TM_USART_PinsPack_2, LL->Baudrate);
    
    /* Initialize reset pin */
    TM_GPIO_Init(GPIOA, GPIO_PIN_0, TM_GPIO_Mode_OUT, TM_GPIO_OType_PP, TM_GPIO_PuPd_UP, TM_GPIO_Speed_Low);
        
    /* We were successful */
    return 0;
}

uint8_t ESP_LL_SendData(ESP_LL_t* LL, const uint8_t* data, uint16_t count) {
    printf("Sending data\n");
	/* Send data */
    TM_USART_Send(USART6, (uint8_t *)data, count);
    
    /* We were successful */
    return 0;
}

uint8_t ESP_LL_SetReset(ESP_LL_t* LL, uint8_t state) {
    /* Set pin according to status */
    if (state == ESP_RESET_SET) {
        TM_GPIO_SetPinLow(GPIOA, GPIO_PIN_0);
    } else {
        TM_GPIO_SetPinHigh(GPIOA, GPIO_PIN_0);
    }
    
    /* We are OK */
    return 0;
}

uint8_t ESP_LL_SetRTS(ESP_LL_t* LL, uint8_t state) {
    /* We are OK */
    return 0;
}

/* USART receive interrupt handler */
void TM_USART6_ReceiveHandler(uint8_t ch) {
	/* Send received character to ESP stack */
	printf("Data received\n");
	ESP_DataReceived(&ch, 1);
}

```

My defines.h has:
```
#define TM_USART6_USE_CUSTOM_IRQ
#define STM32F4xx
#define STM32F469_DISCOVERY
#define ARM_MATH_CM4
#endif
```

MaJerle, 2016-11-08:
Is TM_USART6_ReceiveHandler ever called? If it is, then we have a problem with lib (I never get problem for that actually), if you don't jump to this, there are other problems.

nsalhuana, 2016-11-08:
It is not called from the While Loop in esp8266.c. I do see it called when "AT" is sent and "OK" is received after the while loop times out.

MaJerle, 2016-11-08:
TM_USART6_ReceiveHandler function is called from interrupt (should be, when data received). But it is every called? If you put breakpoint to ESP_DataReceived call in TM_USART6_ReceiveHandler function, do you ever jump there?

nsalhuana, 2016-11-08:
I hit that breakpoint and enter the function ESP_DataReceived. This is only on the "AT" "OK".

I do not see the AT+RST get sent to the ESP8266.

MaJerle, 2016-11-08:
That's because it is not sent. Reset pin is used for device reset.

nsalhuana, 2016-11-08:
Thank you for the help. I hadn't connected that wire in the previous version and was fine. Have a good night!

closed 2016-11-08 by nsalhuana

MaJerle, 2016-11-08:
Does it work now or still doesn't?
Maybe i can consider to send both, AT+RST and pin reset for both situations.

nsalhuana, 2016-11-08:
Yes, it works now and I am able to connect to the network. That would be great. 

Thank you!