MaJerle/ESP8266_AT_Commands_parser #30: espEventConnClosed nor espEventDataReceived get called

nsalhuana, 2016-11-10:
Hi Tilen,

I am using the latest version of your library and am having issues with certain events in ESP_Callback not being called. espEventConnClosed and espEventDataReceived never are called. My TX and RX are set correctly as I am able to initialize the ESP and connect to the Wifi network. Please advise.

Debug Statements on STM32F4:
```
ESP8266 commands parser; Compiled: Nov  9 2016 22:52:38
Initializing module
Initializing ESP
ESP module init successfully!
Connected to network
Connected to example.com!
Connection 0 just became active!
Data sent! Number of bytes sent: 57. We expect connection will be closed by remote server
```

ESP8266 Print out:
```
AT+CIPSEND=0,57

OK
> 
Recv 57 bytes

SEND OK

+IPD,0,1460,93.184.216.34,80:HTTP/1.1 200 OK
Accept-Ranges: bytes
Cache-Control: max-age=604800
Content-Type: text/html
Date: Thu, 10 Nov 2016 05:53:18 GMT
Etag: "359670651"
Expires: Thu, 17 Nov 2016 05:53:18 GMT
Last-Modified: Fri, 09 Aug 2013 23:54:35 GMT
Server: ECS (oxr/837F)
Vary: Accept-Encoding
X-Cache: HIT
x-ec-custom-error: 1
Content-Length: 1270
Connection: close

<!doctype html>
<html>
<head>
    <title>Example Domain</title>

    <meta charset="utf-8" />
    ...
    ...
+IPD,0,7,93.184.216.34,80:in exam
+IPD,0,166,93.184.216.34,80:ples without prior coordination or asking for permission.</p>
    <p><a href="http://www.iana.org/domains/example">More information...</a></p>
</div>
</body>
</html>
0,CLOSED

```

Info:
AT FIrmware: 1.3.0
SDK Version: 2.0.0
Board: STM32F469I-Discovery

Thank you.

MaJerle, 2016-11-10:
Hello,

I just tested again with nucleo board and CLIENT_RTOS example works as expected. Entire output from STM was printed with data received callback.

Did you run CLIENT_RTOS example directly?

nsalhuana, 2016-12-06:
Hi Tilen,

I apologize for the delay in response. I have been traveling to visit customers and have not had time to look at the code.

I haven't been able to run it directly since I am running on OSX and am using Eclipse. I am working on updating an old Windows PC to run Keil.

I have downloaded the latest version of the code that is able to do everything but have the "espEventDataReceived" be called. I will see if I can port it over to Eclipse as that is the IDE of choice. I will post an update. Any tips on porting would be greatly appreciated.

Thanks,

Nicolas

MaJerle, 2016-12-08:
I'm not sure what porting guide you need. C is C and works in C compiler.

nsalhuana, 2016-12-09:
:)

I moved everything over and am still seeing the same issue. I am able to receive some events but I never see the connection go to closed nor data being received. I see the request execute successfully on my FTDI Usb to Serial. I am not sure what else to test. Is it possible that the request is coming in too quickly for the blocking call for send data?

I appreciate the help. I used your previous version of your ESP library and am using your OneWire Library as well.

```
ESP8266 commands parser; Compiled: Dec  8 2016 23:16:46
Initializing ESP
ESP module init successfully!
Connected to network
Connected to example.com!
Connection 0 just became active!
Data sent! Number of bytes sent: 57. We expect connection will be closed by remote server
```

nsalhuana, 2016-12-09:
Hi Tilen,

I was able to get it to receive the data but it seems like it is website dependent. If I use Google.com with an SSL connection, I am able to see that I am receiving data on the STM. If I use Firebase, the data is not sent to the STM. Is this the same parsing issue that we were seeing in the previous version of the parser?

Google.com
```
ESP8266 commands parser; Compiled: Dec  9 2016 00:15:41
Initializing ESP
ESP module init successfully!
Connected to network
Connected to example.com!
Data sent! Number of bytes sent: 60. We expect connection will be closed by remote server
Connection 0 just became active!
1387 bytes of data received on connection 0
1387 bytes of data received on connection 0
1387 bytes of data received on connection 0
1387 bytes of data received on connection 0
1387 bytes of data received on connection 0
138 bytes of data received on connection 0
168 bytes of data received on connection 0
```

Firebase (on ST board):
```
ESP8266 commands parser; Compiled: Dec  9 2016 00:13:52
Initializing ESP
ESP module init successfully!
Connected to network
Connected to example.com!
Data sent! Number of bytes sent: 87. We expect connection will be closed by remote server
Connection 0 just became active!
```
Firebase (on ESP module):
```
+IPD,0,287,104.197.250.13,443:HTTP/1.1 200 OK
Server: nginx
Date: Fri, 09 Dec 2016 07:14:41 GMT
Content-Type: application/json; charset=utf-8
Content-Length: 27
Connection: close
Access-Control-Allow-Origin: *
Cache-Control: no-cache
Strict-Transport-Security: max-age=31556926; includeSubDomains; preload

+IPD,0,27,104.197.250.13,443:{"nics_dev":{"curTemp":69}}0,CLOSED
```

MaJerle, 2016-12-09:
There is no parsing issue anymore, at least should not be as now is processed byte by byte and now it should work ok. For me, I always get response.

Give me your exact GET request (what you send to server) so I can test it by myself too.
Maybe your STM is too slow? Not configured ok (PLL)? Try to run exactly my example or so.

Contact me on email if you still get this issue and will see what can do.

nsalhuana, 2016-12-13:
Hi,

I ran the example.com request and still saw the same issue. I have modified my PLL but am still seeing the same issue.

Were you able to test the request I sent over email?

Thanks,

Nicolas

MaJerle, 2016-12-13:
I was not able to try it yet.
I'm sure code works, problem must be on your side.

Maybe with too small buffer size?

MaJerle, 2016-12-13:
I have tested again and result is positive.

Here are results:
ESP output: http://pastebin.com/fxWQADL9
STM output: http://pastebin.com/6cBEAzhV

I ran my client example directly.

nsalhuana, 2016-12-14:
I truly appreciate all the help. 

I have your OneWire and your previous ESP library all working but I am still not able to get this one to work. 

I have a couple questions for clarification to see if I am missing anything.
What is in your defines.h? Just the "#define TM_USART1_USE_CUSTOM_IRQ"?
What are your PLL settings set to? What do your compile and linker lines look like?

GCC - -mcpu=cortex-m4 -mthumb -mfloat-abi=hard -mfpu=fpv4-sp-d16 -DSTM32F469I_DISCO -DSTM32F4 -DSTM32F469NIHx -DSTM32 -DDEBUG -DUSE_HAL_DRIVER -DSTM32F469xx -DUSE_RTOS_SYSTICK

Linker - -mcpu=cortex-m4 -mthumb -mfloat-abi=hard -mfpu=fpv4-sp-d16 -specs=nano.specs -specs=rdimon.specs -lc -lrdimon

MaJerle, 2016-12-14:
I'm running exactly examples on github. Directly open and compile. You can check entire code there.

I'm not using GCC, but ARM MDK.

nsalhuana, 2016-12-15:
Hi Tilen,

I appreciate all the help. I am FINALLY able to get it working. The issue arose from using semihosting for printfs which slowed down the system just enough to miss the data coming in from certain sites. I am now collecting all the data and printing after the connection closes.

Thank you,

Nicolas

closed 2016-12-15 by nsalhuana