MaJerle/ESP8266_AT_Commands_parser #33: Some warnings

andre117, 2017-01-02:
Thank you for your fine code !

I get the following (minor) issues/compiler warnings:

```
/* Converts number to string */
estatic
void NumberToString(char* str, uint32_t number) {
-->    sprintf(str, "%d", number);
}
```
Remark[Pe181]: argument is incompatible with corresponding format string conversion esp8266.c 583 

/********/

```
/* Process received character */
estatic
--> void ParseReceived(evol ESP_t* ESP, Received_t* Received) {
```
Remark[Pe2349]: declaration hides variable "Received" (declared at line 280) esp8266.c 607

`--> ESP->IPD.Conn->DataStartTime = ESP->Time;       /* Set time when first IPD received on connection */`
Warning[Pa082]: undefined behavior: the order of volatile accesses is undefined in this statement esp8266.c 799

Same warning (Pa082) for the followoing lines
801 (ESP->IPD.Conn->TotalBytesReceived...)
1205 
1277 
1797 
1815 
1828 
1976 

one minor thing:
GSM_RTOS and GSM_ASYNC seem to be never used and explained.

Thank you

MaJerle, 2017-01-02:
I don't understand half of these warnings. What compiler you use?

andre117, 2017-01-02:
IAR Embedded Workbench RX for embedded Renesas RX controllers. See www.iar.com.

Documentation of these issues here
https://www.iar.com/support/user-guides/user-guides-iar-embedded-workbench-for-renesas-rx/

MaJerle, 2017-01-02:
I suggest you check those warnings as you are familiar with IAR and make a pull request to merge changes.

andre117, 2017-01-02:
The only warning that is not easy to fix is the Pa082, which is explained here:

https://www.iar.com/support/tech-notes/compiler/warningpa082-undefined-behavior-the-order-of-volatile-accesses-is-undefined-in-this-statement/

I do not fully understand all ESP->accesses in your code. But the compiler warns that the **order** of those volatile accesses is not defined (C does not define it) and unpredictable behaviour may happen.

andre117, 2017-01-02:
Line 583: change to sprintf( str, "%u", number );
Line 607: Rename variable Received to p_Received to not hide the existing variable
Should GSM_RTOS and GSM_ASYNC mean ESP_RTOS and ESP_ASYNC ?

MaJerle, 2017-01-02:
Hiding variable is valid in C, don't know why they throw warnings, anyway, I fixed.

GSM_RTOS and GSM_ASYNC will be fixed by me.