MaJerle/ESP8266_AT_Commands_parser #34: espEventConnClosed callback parameter always empty

matteofumagalli1275, 2017-01-12:
Hi,
noticed that when i get the espEventConnClosed event the connection handler has already been cleared.

For now i moved che connection clear after the callback call:
```
#if !ESP_SINGLE_CONN
    if (strncmp(&str[1], FROMMEM(",CONNECT"), 8) == 0) {
        ESP_CONN_t* conn = (ESP_CONN_t *)&ESP->Conn[CHARTONUM(str[0])]; /* Get connection from number */
        conn->Number = CHARTONUM(str[0]);                   /* Set connection number */
        conn->Flags.F.Active = 1;                           /* Connection is active */
        conn->Callback.F.Connect = 1;
    } else if (strncmp(&str[1], FROMMEM(",CLOSED"), 7) == 0) {
        ESP_CONN_t* conn = (ESP_CONN_t *)&ESP->Conn[CHARTONUM(str[0])]; /* Get connection from number */
        //__CONN_RESET(conn);                                 /* EDIT Reset connection */
        conn->Callback.F.Closed = 1;
    }
#else
```
```
 if (__IS_READY(ESP) && c->Callback.F.Closed) {      /* Connection just closed */
     c->Callback.F.Closed = 0;
     ESP->CallbackParams.CP1 = c;
     ESP_CALL_CALLBACK(ESP, espEventConnClosed);
     __CONN_RESET(c); //EDIT
 }
```

MaJerle, 2017-01-12:
Is this a problem?
I notify use that connection is closed. That's all. From here, you are done.
I'm not sure what is actual problem here, maybe you can explain the details why is this wrong.

matteofumagalli1275, 2017-01-12:
Because i have multiple connections i find useful to get informations about which connection was closed.
Also i saw that you send the connection handler with the event but is always empty, so i though that wasn't the expected behaviour.

MaJerle, 2017-01-12:
Ok, having informations might be useful.

> Also i saw that you send the connection handler with the event but is always empty, so i though that wasn't the expected behaviour.

Can you explain this?

matteofumagalli1275, 2017-01-12:
Yeah, reading the code i saw this:

```
 if (__IS_READY(ESP) && c->Callback.F.Closed) {      /* Connection just closed */
     c->Callback.F.Closed = 0;
     ESP->CallbackParams.CP1 = c;
     ESP_CALL_CALLBACK(ESP, espEventConnClosed);
 }
```
but 'c' is always empty because you clear that before setting Callback.F.Closed to 1.

```
else if (strncmp(&str[1], FROMMEM(",CLOSED"), 7) == 0) {
        ESP_CONN_t* conn = (ESP_CONN_t *)&ESP->Conn[CHARTONUM(str[0])]; /* Get connection from number */
        __CONN_RESET(conn);                                 /*  Reset connection */
        conn->Callback.F.Closed = 1;
}
```
see __CONN_RESET.

Did i misunderstood something?

MaJerle, 2017-01-12:
Pointer is still valid so in your callback if you do if (closedConn == mystartedConn) ...

Btw..I've commited new code.

matteofumagalli1275, 2017-01-12:
That's true, didn't thought about it.

I pulled yesterday.

MaJerle, 2017-01-12:
After thinking more about this I came to conclusion.
It is actually better to close connection when notification is sent, because that way I don't allow user to create new connection if notification is not sent before. But this breaks if connection became active again between time I set connection to call closed callback and call for callback. In this case, server mode is a problem.

wroom02, 2017-05-03:
Hello,
I have the same problem.
I open a few Sockets (sock1, sock2, sock3, sock4) on the client mode. The server closes the socket sequentially one after the other. In the callback function I read number of lockable slot. Unfortunately, this value is the same for all sockets. This results in paralysis of the machine. 
Is this a bug in the library or incorrect implementation?
I will be grateful for your help. 
Best regards Adam

MaJerle, 2017-05-04:
According to the code, you should have different connections.