MaJerle/ESP8266_AT_Commands_parser #36: Confusion with connection

gadag99, 2017-02-03:
Hi Majerle, 

I am using STM32F04VG and esp8266-01 , In pin configuration table you told that CTS = PA3, But in esp8266-01 CTS pin is not available, please suggest me what is the alternate way.

MaJerle, 2017-02-03:
I have also mention that you can disable this feature.

gadag99, 2017-02-03:
Ok, i tried with that also , I dint not connect that pin. when i run the program (UART_WIFI_RTOS), I am getting errors like  
> ESP Init error. Status: 1 ,
> Problems trying to connect to network: 1 
 My AT Version is 1.5.0 , Weather i need ult-rate in the program ? or its should work ?

MaJerle, 2017-02-03:
If you have disabled CTS feature, then this does not cause a problem.
If you did not setup ESP device properly (CD_PH pin, GPIO0, GPIO2) then it may not work.

Make sure first ESP responds with messages.

Also, make sure TX and RX are connected properly.

gadag99, 2017-02-03:
Thanks for your time, I am getting confusing , Please could you tel me the , GPIO0 and GPIO2  to connect which pin? I shorted Cd_PH and vcc both. I am not connect GPIO0 AND GPIO2

MaJerle, 2017-02-03:
Please use Google for ESP configuration. My library only uses AT commands from ESP.

gadag99, 2017-02-03:
Hi majerle , I am really sorry, I am disturbing u lot and may asking silly questions. but i am get inspired by your awesome work. As you told i gone through  esp8266at your document , as you mentioned i connect the pins GPIO0 and GPIO2 vcc and rest of the pins connects as its is, but still ai m getting same errors. 
CTS pin disable mean is their any thing to modify in program or how? simply i keep open. Please suggest me how to configure with my stm32 board. once again sorry , it may silly for you, i am beginner for this. please help me in this , waiting for your valuable response.

Thanks
gadag

MaJerle, 2017-02-03:
I did not get an anwer from you regarding if device respond to your messages if you control it with USB->UART converter. Start there, then go to STM32.

gadag99, 2017-02-07:
Hi, 
 Does this your programs supports for STM32F407?

MaJerle, 2017-02-07:
Library can be used with any device. Even arduino can do it.

gadag99, 2017-02-07:
ok, I wanted to know about your examples like EXAMPLE_UART_WIFI_RTOS and  EXAMPLE_SERVER_RTOS this examples are not support for stm32f407 ?

MaJerle, 2017-02-07:
Like I said. Any device can be used, even f407.

gadag99, 2017-02-07:
Thank you , it may sound silly, can you please tell ,where and what  changes is to be done to work with stm32F407

closed 2017-05-16 by MaJerle