MaJerle/ESP8266_AT_Commands_parser #37: Program gets stuck on ESP_WaitReady

Snapsion, 2017-02-04:
I'm using the ESP8266 library, on an STM32F4-Discovery board with HAL libraries.

MaJerle, 2017-02-04:
I have to say I don't have crystal bubble to know what is going on. You may write me detailed informations about your setup, your ESP config, is RTOS, non-RTOS, etc.

Your info right now is irrelevant to me.

Snapsion, 2017-02-05:
I'm not using an RTOS. ESP-01 is updated to the latest firmware and connected to UART4.

This is my main code:

    int main(void)
    {
     HAL_Init();
     SystemClock_Config();
     MX_GPIO_Init();
     MX_DMA_Init();
     MX_UART4_Init();
     HAL_UART_Receive_DMA(&huart4,&ch,1);// DMA set to circular mode
     HAL_Delay(1000);
	printf("Program starting..\n");
	
	if(ESP_Init(&ESP8266,115200,ESP_Callback)==espOK)
		printf("ESP8266 successfully initialized :D\r\n");
	else
		printf("An error occurred :(\r\n");

	while(1);
        }

I've set the user button to send AT\r\n and printf in the UART reception callback.

       void HAL_UART_RxCpltCallback(UART_HandleTypeDef *huart) {
	/*uint8_t ch;*/
	/* Get character from USART */
	
	printf("%c",ch);
	/* Send received character to ESP stack */
	ESP_DataReceived(&ch, 1);
        }

The module responded OK, each time I pressed the button.

While debugging the code I noticed that large portions of code weren't compiled(impossible to set breakpoints), including the ESP_LL_SendData as well as a lot of private functions in esp8266.c. I've set the optimization level to Level 0 (I'm using Keil), but to no avail.

MaJerle, 2017-02-05:
Private functions are not part of compile if settings are disabled.

In your case I would say you did not add file to project or you did not include debug informations in keil options.

MaJerle, 2017-02-05:
Rx complete callback is called on each receove char. Sending this char can lead to the fact that you will fail in reading next xhar of rx because your transmission. Printf in rx irq is bad idea.

Snapsion, 2017-02-12:
Hi Tilen,
 After experimenting with various settings, I've traced my issue to the low level send data function.
It seems that when I use transmit with DMA, "\r\n" isn't transmitted, and the module keeps echoing back 'ATATATATAT....' until initialization fails.

 Using polling mode transmission solved my issue. Now module gets initialized successfully.

MaJerle, 2017-02-12:
Logical. Either you wait for dma to finish last request before you start new or you create buffer for send data to handle dma.

closed 2017-02-12 by Snapsion