MaJerle/ESP8266_AT_Commands_parser #38: Issue with OUTPUT

ABHILASH700, 2017-02-06:
Hello Tilen,
I'm Using STM32F407VG board and ESP 8266-01(version-1.5.2) for your UART_WIFI_RTOS example.But wifi module not responding, i'm getting output as follows.

ESP8266 commands parser; Compiled: Feb  6 2017 14:55:00
ESP Init error. Status: 1
Problems trying to connect to network: 1.
I did hardware settings as 
                                         ESP 01->STM32F407
                                          RX-> PA9
                                          TX->PA10
                                          VCC->3.3v
                                          RST->PA0
                                          GND->GND
                                          CTS->Not connected since this pin is not in ESP 01.
                                          GPIO 01, GPIO 02 and  CH_EN->3.3v

In programming i made variation as follows,
I'm using PLL values as M=8,N=336,P=2,Q=7,  HSE value=8000000 and HSI value=16000000.
suggest me where i'm doing wrong? why wifi not responding with stm32f4?

MaJerle, 2017-02-06:
If you diagnose what is transmitted over UART, is anything received from ESP?
Is anything sent to ESP device from STM?

ABHILASH700, 2017-02-06:
Thanks for quick response.
Ya i'm sending AT commands to ESP 01 through UART. In your example shown that we are sending AT command if we get response=espOK then we will get output as ESP module init successfully.
Except PLL setting i didn't changed anything on your program.

MaJerle, 2017-02-06:
Follow my questions. If you inspect UART data lines, do you see actual AT\r\n command sent to device?
Do you see answer on the line? 

Use USB->UART converter and check data on terminal.

ABHILASH700, 2017-02-06:
Hey Tilen,

I followed your instruction, But i'm receiving on AT\r\n data on terminal. What i did mistake in that example?

MaJerle, 2017-02-06:
Show me your initialization code you get from ESP.

ABHILASH700, 2017-02-06:
`/**
 * Keil project example for ESP8266 CLIENT mode (UART WIFI passthrough) and RTOS support
 *
 * @note      Check defines.h file for configuration settings!
 * @note      When using Nucleo F411 board, example has set 8MHz external HSE clock!
 *
 * Before you start, select your target, on the right of the "Load" button
 *
 * @author    Tilen Majerle
 * @email     tilen@majerle.eu
 * @website   http://stm32f4-discovery.net
 * @ide       Keil uVision 5
 * @conf      PLL parameters are set in "Options for Target" -> "C/C++" -> "Defines"
 * @packs     STM32F4xx/STM32F7xx Keil packs are requred with HAL driver support
 * @stdperiph STM32F4xx/STM32F7xx HAL drivers required
 *
 * \par Description
 *
 * This examples shows how you can use ESP for UART to WiFi bridge as client device
 *
 * Requirements:
 * - Create listening socket on your PC (or any other server with this support)
 * - For Windows, a great tool is TCP/IP Builder, where you can create socket and listen to it.
 * - ESP stack must be in single connection mode. Check esp8266_config.h file for mode informations.
 *
 * - Library is initialized using ESP_Init
 * - Device must connect to network. Check WIFINAME and WIFIPASS defines for proper settings for your wifi network
 * - Set IP address for your PC and port where PC listens for your socket
 * - On button press, device will go to UART-WIFI mode, send some data and wait user to press button again to close UART-WIFI mode
 * - Connection will close
 *
 * \note  Example uses single buffer for all connections.
 *        Since only one connection is active at a time in this example, there is no problems with that.
 *
 * \par Pinout for example (Nucleo STM32F411)
 *
\verbatim
ESP         STM32F4xx           DESCRIPTION
 
RX          PA9                 TX from STM to RX from ESP
TX          PA10                RX from STM to RX from ESP
VCC         3.3V                Use external 3.3V regulator
GND         GND
RST         PA0                 Reset pin for ESP
CTS         PA3                 RTS from ST to CTS from ESP
            BUTTON(PA0, PC13)   Discovery/Nucleo button, depends on configuration
            
            PA2                 TX for debug purpose (connect to PC) with 921600 bauds
\endverbatim
 */
/* Include core modules */
#include "stm32fxxx_hal.h"
/* Include my libraries here */
#include "defines.h"
#include "tm_stm32_disco.h"
#include "tm_stm32_delay.h"
#include "tm_stm32_usart.h"
#include "esp8266.h"
#include "cmsis_os.h"

#define DEBUG_USART         USART2
#define DEBUG_USART_PP      TM_USART_PinsPack_1

/* Wifi network settings, replace with your settings */
#define WIFINAME            "pixzee2"
#define WIFIPASS            "pixzeE*28"

/* Set up your server settings to connect to */
#define IP_ADDR             "192.168.0.106"
#define PORT                100
[... 166 lines cut ...]
            ESP_CONN_t* conn = (ESP_CONN_t *)params->CP1;   /* Get connection for event */
            printf("Connection %d was just closed!\r\n", conn->Number);
            
            break;
        }
        case espEventTransparentReceived: {                 /* This event is always called from thread where ESP_Update function is called */
            uint8_t ch = *(uint8_t *)params->CP1;           /* Get single character from device */
            printf("%c", ch);                               /* Print character to screen */
            
            break;
        }
        default:
            break;
    }
    
    return 0;
}

/* printf handler */
int fputc(int ch, FILE* fil) {
    TM_USART_Putc(DEBUG_USART, ch);         /* Send over debug USART */
    return ch;                              /* Return OK */
}
`
I'm using this main code.

MaJerle, 2017-02-06:
You obviously don't know what I'm talking to you.

STM sends AT commands to device and device responds. I would like to know if STM sends anything on UART data lines and if ESP responds with aanything.

And I wanna see what is this actual response from ESP.

Don't play stupid with me, please.

ABHILASH700, 2017-02-06:
Sorry Tilen.
Since i'm beginner in this field so i didn't got your question. I thought by following your examples we directly get output just by setting PLL in programming. Please give me guidance to understand this problem.

closed 2017-05-16 by MaJerle

Ksenakis, 2017-05-27:
Hello! I have the same problem, stm sends AT + PCT, and in return receives

AT + RST

OK

  Ets Jan 8 2013, rst cause: 1, boot mode: (3,6)

Load 0x40100000, len 1396, room 16
Tail 4
Chksum 0x89
Load 0x3ffe8000, len 776, room 4
Tail 4
Chksum 0xe8
Load 0x3ffe8308, len 540, room 4
Tail 8
Chkum 0xc0
Csum 0xc0

2nd boot version: 1.4 (b1)
   SPI Speed: 40MHz
   SPI Mode: DIO
   SPI Flash Size & Map: 8Mbit (512KB + 512KB)
Jump to run user1 @ 1000

Rl $? Rlv
Ai-Thinker Technology Co., Ltd.

Invalid

Ksenakis, 2017-06-15:
Hello! I have the same problem, stm sends AT + RST, and in return receives

AT + RST

OK

  Ets Jan 8 2013, rst cause: 1, boot mode: (3,6)

Load 0x40100000, len 1396, room 16
Tail 4
Chksum 0x89
Load 0x3ffe8000, len 776, room 4
Tail 4
Chksum 0xe8
Load 0x3ffe8308, len 540, room 4
Tail 8
Chkum 0xc0
Csum 0xc0

2nd boot version: 1.4 (b1)
   SPI Speed: 40MHz
   SPI Mode: DIO
   SPI Flash Size & Map: 8Mbit (512KB + 512KB)
Jump to run user1 @ 1000

Rl $? Rlv
Ai-Thinker Technology Co., Ltd.

Invalid

MaJerle, 2017-06-15:
The code does not continue because there is missing *ready* response from ESP.

Ksenakis, 2017-06-19:
Hello! Changed ESP, eventually ESP began to respond to requests. But the connection does not work

ESP8266 commands parser; Compiled: Jun 19 2017 21:15:08

ERROR
AT+RST

OK

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 1856, room 16 
tail 0
chksum 0x63
load 0x3ffe8000, len 776, room 8 
tail 0
chksum 0x02
load 0x3ffe8310, len 552, room 8 
tail 0
chksum 0x79
csum 0x79

2nd boot version : 1.5
  SPI Speed      : 40MHz
  SPI Mode       : DIO
  SPI Flash Size & Map: 8Mbit(512KB+512KB)
jump to run user1 @ 1000

ЊrlДlдЗl`ЊrlЗlдl`Њrl$дll`rl$rlЊbblЊbr|bpbbbтn'nn‚lllмДlЊllдДЗдl$`ЂnЊbblмЊbЊ’bblЊbprlrl
Ai-Thinker Technology Co. Ltd.

ready
AT

OK
ATE0

OK
AT version:1.1.0.0(May 11 2016 18:09:56)
SDK version:1.5.4(baaeaebb)
Ai-Thinker Technology Co. Ltd.
Jun 13 2016 11:29:20
OK

OK

OK

OK
+CIPSTAMAC_CUR:"60:01:94:1c:d9:33"

OK
+CIPAPMAC_CUR:"62:01:94:1c:d9:33"

OK
+CIPSTA_CUR:ip:"0.0.0.0"
+CIPSTA_CUR:gateway:"0.0.0.0"
+CIPSTA_CUR:netmask:"0.0.0.0"

OK
+CIPAP_CUR:ip:"192.168.4.1"
+CIPAP_CUR:gateway:"192.168.4.1"
+CIPAP_CUR:netmask:"255.255.255.0"

OK
+CWSAP_CUR:"AI-THINKER_1CD933","",1,0,4,0

OK
no change

OK

ERROR

busy p...

busy p...

busy p...
WIFI CONNECTED
WIFI GOT IP

OK