MaJerle/ESP8266_AT_Commands_parser #41: Can't handle button press

suvinh, 2017-04-11:
Hi Tilen.

I'm trying to test your source 01-EXAMPLE_CLIENT_PT on STM32F429 discovery with module ESP8266 v1 and external power as suggestion.
I can't handle button press, it always set the variable "client" to 1 right after entering the loop "while(1)".
After running the first time, the variable client can't be set anymore.

I put the toogle led code like below:
if (TM_DISCO_ButtonOnPressed()) {                   /* Handle button press */
      client = 1;
      TM_DISCO_LedToggle(LED_GREEN);
}
Then the green LED just toggle 1 time.

If I comment all the esp8266 functions, the button function works normally.
I also tested with the source 01-EXAMPLE_CLIENT_RTOS, but it behaves as so, too.
Did I do something wrong? Can you tell me how to make it work well?

Thank you very much for you help.

MaJerle, 2017-04-11:
Dont see which board you use.

MaJerle, 2017-04-11:
Ok. Did you set your board when compiling? Did you correct uart setup?

suvinh, 2017-04-11:
Yes, I did.
The log is like below.

"ESP8266 commands parser; Compiled: Apr 11 2017 15:37:56
ESP module init successfully!
Connected to network
Start to test button
Connection to thingspeak.com has started!
Stack is IDLE!
Connection 0 just became active!
Connection to thingspeak.com has been successfull!
Data sending has started successfully
Stack is IDLE!
Data sent! Number of bytes sent: 44. We expect connection will be closed by remote server
66 bytes of data received on connection 0
Connection 0 was just closed!"

Just one time, then the green LED is always on, pressing the button has no effect.

MaJerle, 2017-04-11:
Ok, so lib works for you, which sounds great!

Please check which pin is used for button **in software** and check if it is PA0 or is anything else.

suvinh, 2017-04-11:
I changed the button pin to pin G7, then everything works well.
When using the debugging, I saw that the value at pin A0 was always set to 1.
But if I put all esp8266 functions in comment, I could read the value at A0 normally.
I'm a little confusing.
Can you guess what's happening with the value at pin A0? I wanna understand this.

I configured gpio directly from tm_gpio file, not tm_discovery file.

"TM_GPIO_Init(GPIOG, GPIO_PIN_7, TM_GPIO_Mode_IN, TM_GPIO_OType_PP, TM_GPIO_PuPd_UP, TM_GPIO_Speed_Low);
//TM_GPIO_Init(GPIOA, GPIO_PIN_0, TM_GPIO_Mode_IN, TM_GPIO_OType_PP, TM_GPIO_PuPd_DOWN, TM_GPIO_Speed_Low);"

MaJerle, 2017-04-11:
Check esp lowlevel part of code.

suvinh, 2017-04-11:
Ouh. I understand why.
Thankyou very much for your help and your lib. ^^
Nice day.

closed 2017-04-11 by MaJerle