MaJerle/ESP8266_AT_Commands_parser #43: ESP32 Command Incompatibility

schwac, 2017-04-18:
Hi, I noticed that the ESP32 is not compatible with this library due to the use of _CUR and _DEF appended to the ends of some of the commands. Is there a plan to make this library compatible? It could probably be done pretty easily via a #define for ESP type? 

Thanks!

MaJerle, 2017-04-18:
Hi.

This lib uses _CUR and _DEF on commands where this should be used. But I don't know if ESP32 has AT commands? Does it? If so, please give me a link where I can see more about.

schwac, 2017-04-18:
Here is the link: https://espressif.com/sites/default/files/documentation/esp32_at_instruction_set_and_examples_en.pdf

It is not available as a binary yet, but the AT firmware is in a repo here:
https://github.com/espressif/esp32-at

It is pretty easy to build the firmware using their make environment and step by step instructions. I actually built a binary version and posted it here:
https://esp32.com/viewtopic.php?f=2&t=1103&start=50

Also, I think the difference is that the ESP32 doesn't have the _CUR and _DEF, whereas the ESP8266 does. I would hugely appreciate if you could get this working! This library is awesome!

MaJerle, 2017-04-18:
Looks nice. Using some macros I can get this to work. Can you prepare working binary file so I can easily upload to ESP?

schwac, 2017-04-18:
Do you have an ESP32? The binaries are attached. 
[ESP32_AT_FW_v0.1.zip](https://github.com/MaJerle/ESP8266_AT_Commands_parser/files/930609/ESP32_AT_FW_v0.1.zip)

This version uses UART0, RX pin 3, TX pin 1, and disables log data on UART0 as well. It can be loaded with the ESP Flash Download Tools utility:
https://espressif.com/en/support/download/other-tools

MaJerle, 2017-04-18:
I don't. I just ordered one :)

schwac, 2017-04-18:
SPI would be cool, or increase the UART baud rate during init?

MaJerle, 2017-04-18:
The idea of spi is that module notifies you about data and you read it when you want it. With fastest possible speed uart cant handle. If possible in module of course.

Notification can be done with irq pin from module to device.

schwac, 2017-04-18:
The way you can make it work with UART is to use a DMA Rx circular buffer (I'm using STM32F4 as the main processor) to ensure data can be received at any speed, any time. I have some custom boards designed with a UART connection, so I have a personal interest in making it work!

MaJerle, 2017-04-18:
I got your point. Just you need ram. For F4 is not a problem. First idea of this lib was suppirt for low ram. 

If you want you can modify and add pull request. Im little bit without time. When i got module I can try. Must use defines for enabling esp32.

schwac, 2017-04-18:
I'll give the changes a shot when I get a chance in the next few weeks.