MaJerle/ESP8266_AT_Commands_parser #45: yield macro for RTOS in tickless mode

d21d3q, 2017-05-12:
I am playing around with running this library on chibios which on default runs in tickless mode. It means, that context doesn't switch in regular period of time, but only if task finishes or enters sleep or yields. 
`ESP_WaitReady` in RTOS mode doesn't rung `ESP_Update` which is supposed to run in separate thread so that it tickless mode it will block in while loop. In order to make it working I had to place there yield statement. 
I think that there should be some macro defined in config file which allows user to place yield statement wherever library may block waiting for other thread.

MaJerle, 2017-05-12:
This is a good point and has meaning to add it.

MaJerle, 2017-05-13:
Since your point was good to add, I decided to completelly rewrite low-level part to use only 1 function to ease features in future. Please take a look at least at example, and implement yeild macro on your needs.

After that, I will review code and merge together.

d21d3q, 2017-05-13:
What kind of ease will bring single low-level function? (how is it related to yielding?)

MaJerle, 2017-05-13:
Instead of new funcrion for this for compatibility issues in future it is better to use single function which is already implemented by user with diff parameters. Then user decides to react on specific command if desired or not.

d21d3q, 2017-05-14:
For me void pointers don't seem to be compatibility friendly. If you decide to change type of parameter, and user won't notice that and will keep casting pointers to previous type, he will end up with debugging bad pointers... 
If the case is to not force user to define all of callbacks, then they can be defined as empty weak functions (code would compile but in case of changing parameter type users callback won't be linked), but I wouldn't go for that and keep previous implementation (and here compiler will produce error and fore user to reimplement functions) - since this is low-level part of library needed for proper working, I would require from user to know how it works and be aware of what will not work if he don't implement it. For example I am having `ESP_SYS_Create` and `ESP_SYS_Delete` as empty functions because I am declaring mutex with macro in global scope. 
By the way, in your examples you are using CMSIS mutexex with FreeRTOS. When I tried to use [mutex from FreeRTOS API](http://www.freertos.org/CreateMutex.html), then program was hanging up [here](https://github.com/MaJerle/ESP8266_AT_Commands_parser/blob/master/00-STM32_LIBRARIES/FreeRTOS/tasks.c#L3466) because system requires that only thread which took mutex can release it. (So that I had to use [semaphore](http://www.freertos.org/xSemaphoreCreateBinary.html) 
About yielding, I will implement it on Tuesday.

MaJerle, 2017-05-14:
Regarding locking on one thread and unlocking in another, I'm aware of it.

Regarding weak parameter. Keep in mind, weak parameter is no C standard and implementation is important for specific compiler. I'm not saying void pointers are the best, but they give you flexibility over different types of parameters. There is documentation on what parameter is used for "param" and what for "result", so there should be no problem as users can see easily.

When new release is out, you will be aware of low-level data change if there will be any, which I don't believe in fact.

closed 2017-05-16 by MaJerle