MaJerle/ESP8266_AT_Commands_parser #47: Library with FreeRTOS

wine18, 2017-05-16:
Hi,

I am trying to run a project on STM32F429, when my device will try to connect to TCP server on PC. I run project on my own setup with FreeRTOS. I used your code from master, #50e3e08 and ported 1 of your examples to send some basic short strings ( 9 bytes). I do it every time I click a button, so it is maximum 2 times per second. But when I observe my logs, there is a problem will callbacks - the callback for espEventDataSent is not called everytime, when the data is sent. Can you tell me, is this proper way of use of your library, or I am doing something wrong?  Most of the task code I pasted below with configuration file:

```
#define ESP_BUFFER_SIZE                     1024
#define ESP_SINGLE_CONN                     1
#define ESP_CONNBUFFER_SIZE                 5840
#define ESP_CONN_SINGLEBUFFER               0
#define ESP_ECHO                            1
#define ESP_RTOS                            1
#define ESP_ASYNC                           0

```

```
Thread1()
{
    ESP_Init(&internals.ESP, 115200, ESP_Callback);
    connect_to_router(); // calls ESP_STA_Connect(&internals.ESP, internals.wifi_name, internals.wifi_pass, NULL, 0, 1)
    while(1)
    {
        ESP_ProcessCallbacks(&internals.ESP);

        if(BSP_Blue_Button_GetState(BSP_BUTTON_0))   /* Handle button press */
        {
            /* Try to connect to server as client, connect to 192.168.0.33:4321 */
            if(ESP_open_connection()) //calls ESP_CONN_Start(&internals.ESP, &internals.conn, ESP_CONN_Type_TCP, internals.conn_target_ip,
            internals.conn_target_port, 1)
            {
                while(1)
                {
                    if(BSP_Blue_Button_GetState(BSP_BUTTON_0))   /* Handle button press */
                    {
                        /* Send HTTP request for example */
                        if ((internals.espRes = ESP_CONN_Send(&internals.ESP, internals.conn, requestData, sizeof(requestData), &internals.bw, 1)) == espOK)
                        {
                            WIFI_DEBUG("Data sent! Number of bytes sent: %d\r\n", internals.bw);
                        }
                    }
                    vTaskDelay(1);
                }
            }
            else
            {
                WIFI_DEBUG("Problems trying to connect to server as client: %d\r\n", internals.espRes);
            }
            ESP_close_connection();
        }
        vTaskDelay(1);
    }
}
```

```
void ESP_Update_Thread(void *p)
{
    while (1)
    {
        ESP_Update(&internals.ESP);
    }
}
```

```
int ESP_Callback(ESP_Event_t evt, ESP_EventParams_t* params)
{
    switch (evt)
    {   /* Check events */
    case espEventIdle:
        WIFI_DEBUG("Stack is IDLE!\r\n");
        break;
    case espEventConnClosed: {
        ESP_CONN_t* conn = (ESP_CONN_t *)params->CP1;   /* Get connection for event */
        WIFI_DEBUG("Connection %d was just closed, retrying to connect...\r\n", conn->Number);
        internals.is_tcp_connected = false;
        ESP_open_connection();
        break;
    }
    case espEventDataSent:
        internals.conn = (ESP_CONN_t *)params->CP1;   /* Get connection for event */
        WIFI_DEBUG("Data sent on conn: %d\r\n", internals.conn->Number);
        break;
    }
    default:
        break;
    }
    return 0;
}
```

The result looks like this:

> [DBG WIFI] Data sent! Number of bytes sent: 9
> [DBG WIFI] Data sent on conn: 0
> [DBG WIFI] Data sent! Number of bytes sent: 9
> [DBG WIFI] Data sent! Number of bytes sent: 9
> [DBG WIFI] Data sent! Number of bytes sent: 9
> [DBG WIFI] Data sent on conn: 0
> [DBG WIFI] Data sent! Number of bytes sent: 9
> [DBG WIFI] Data sent on conn: 0
> [DBG WIFI] Data sent! Number of bytes sent: 9
> [DBG WIFI] Data sent! Number of bytes sent: 9
> [DBG WIFI] Data sent on conn: 0
> [DBG WIFI] Data sent! Number of bytes sent: 9
> [DBG WIFI] Data sent on conn: 0
> [DBG WIFI] Data sent! Number of bytes sent: 9
> [DBG WIFI] Data sent on conn: 0
> [DBG WIFI] Data sent! Number of bytes sent: 9
> [DBG WIFI] Data sent on conn: 0

Another question, if it wouldn't be RTOS project, on which ISR should be called ESP_Update() function? And right now, should it be in separate task, without any vTaskDelay() with it?

MaJerle, 2017-05-16:
Of course it is not called because you are in `while (1)` loop inside function. I don't see where you jump out of inner `while (1)` loop. Callbacks are called from `ESP_ProcessCallbacks` function.

wine18, 2017-05-16:
Every Library Call I should use `ESP_ProcessCallbacks` ?
Ok, I thought that is the purpose of the `ESP_Update()` function.

But what with my last question:

> Another question, if it wouldn't be RTOS project, on which ISR should be called ESP_Update() function? And right now, should it be in separate task, without any vTaskDelay() with it?

Thank you for quick response!

MaJerle, 2017-05-16:
You should call `ESP_ProcessCallbacks` as fast as possible, usually from the same thread as main app. In general, if you used blocking calls and return from `_Send` function was `espOK`, then data were sent and there is no need for callbacks.

You should do it either in separate periodical interrupts (SYSTICK for example) or call it from main while loop (I prefer this). But then you have to be aware of blocking calls as it may block your device. Check implementation of my examples using protothreads.

wine18, 2017-05-16:
Ok, I wasn't aware of that. In my project it will be necessary to handle returning data from server. To make it properly work, I need to call `ESP_ProcessCallbacks` as fast as I can to not lose any data? Because right now, I am not sure, how I should receive data.  I prepared separate callback:
```
    case espEventDataReceived: {
        ESP_CONN_t* conn = (ESP_CONN_t *)params->CP1;   /* Get connection for event */
        uint8_t* data = (uint8_t *)params->CP2;         /* Get actual received data */
        uint16_t datalen = params->UI;                  /* Print length of received data in current part packet */
        (void)(data);                                   /* Prevent compiler warnings */

        WIFI_DEBUG("%d bytes of data received on connection %d\r\n", datalen, conn->Number);
        break;
    }
```

but it is never called - I am sending from the server 6 bytes of data. Even, when I added `ESP_ProcessCallbacks(&internals.ESP);` in the inner while(1) loop...

wine18, 2017-05-17:
I was able to run one of example code "EXAMPLE_UART_WIFI_RTOS" with code from #664ec74 and I am able to send data to the server, but received data from server is filled with zeroes...

The debugs are here:

[image]

The logs from STM32F429Discovery Rx line looks like this:
```
[noise after reset]
ready{h0D}
AT{h0D}{h0D}
{h0D}
OK{h0D}
ATE0{h0D}{h0D}
{h0D}
OK{h0D}
AT version:1.1.0.0(May 11 2016 18:09:56){h0D}
SDK version:1.5.4(baaeaebb){h0D}
compile time:May 20 2016 15:06:44{h0D}
OK{h0D}
{h0D}
OK{h0D}
{h0D}
OK{h0D}
{h0D}
OK{h0D}
+CIPSTAMAC_CUR:"5c:cf:7f:13:f6:25"{h0D}
{h0D}
OK{h0D}
+CIPAPMAC_CUR:"5e:cf:7f:13:f6:25"{h0D}
{h0D}
OK{h0D}
+CIPSTA_CUR:ip:"0.0.0.0"{h0D}
+CIPSTA_CUR:gateway:"0.0.0.0"{h0D}
+CIPSTA_CUR:netmask:"0.0.0.0"{h0D}
{h0D}
OK{h0D}
+CIPAP_CUR:ip:"192.168.4.1"{h0D}
+CIPAP_CUR:gateway:"192.168.4.1"{h0D}
+CIPAP_CUR:netmask:"255.255.255.0"{h0D}
{h0D}
OK{h0D}
+CWSAP_CUR:"ESP_13F625","",1,0,4,0{h0D}
{h0D}
OK{h0D}
{h0D}
ERROR{h0D}
WIFI CONNECTED{h0D}
WIFI GOT IP{h0D}
{h0D}
OK{h0D}
+CIPSTA_CUR:ip:"192.168.0.69"{h0D}
+CIPSTA_CUR:gateway:"192.168.0.1"{h0D}
+CIPSTA_CUR:netmask:"255.255.255.0"{h0D}
{h0D}
OK{h0D}
{h0D}
OK{h0D}
STATUS:2{h0D}
{h0D}
OK{h0D}
CONNECT{h0D}
{h0D}
OK{h0D}
STATUS:3{h0D}
+CIPSTATUS:0,"TCP","192.168.0.33",100,49781,0{h0D}
{h0D}
OK{h0D}
{h0D}
OK{h0D}
{h0D}
>aaaaaaaaaaaaaaaaaaaaaaaaaaZZZZZZZZZZZZZZ
```

Any ideas what can be wrong? Bad version of AT commands in ESP or wrong revision of library? If yes, could you suggest which one is prefered?

MaJerle, 2017-05-18:
I don't know what is this "aaaaaaaaaaaaaaaaaaaaaaaaaaZZZZZZZZZZZZZZ" in your part. Do you send this?

wine18, 2017-05-18:
Yes, I sending byte by byte. 1 byte at a time and this is the whole string I sent from the server. I am using Herkules terminal (http://www.hw-group.com/products/hercules/index_en.html) to setUp the server.