MaJerle/ESP8266_AT_Commands_parser #50: AT+CIPSEND returns an error

lavruhin, 2017-06-09:
Hi!

I'm using your library.
Almost every time an attempt to send data via the ESP module (CIPSEND command) ends with an error.
Very rarely CIPSEND ended successfully.
What could be the cause of the error?
Here are the exchange data with the module:

AT
AT

OK
AT+GMR
AT+GMR
AT version:1.3.0.0(Jul 14 2016 18:54:01)
SDK version:2.0.0(656edbf)
compile time:Oct 29 2016 22:56:21
OK
AT+CWMODE=3
AT+CWMODE=3

OK
AT+CIPMUX=1
AT+CIPMUX=1

OK
AT+CIPSTAMAC_CUR?
AT+CIPSTAMAC_CUR?
+CIPSTAMAC_CUR:"18:fe:34:a0:59:1d"

OK
AT+CIPAPMAC_CUR?
AT+CIPAPMAC_CUR?
+CIPAPMAC_CUR:"1a:fe:34:a0:59:1d"

OK
AT+CIPSTA_CUR?
AT+CIPSTA_CUR?
+CIPSTA_CUR:ip:"0.0.0.0"
+CIPSTA_CUR:gateway:"0.0.0.0"
+CIPSTA_CUR:netmask:"0.0.0.0"

OK
AT+CIPAP_CUR?
AT+CIPAP_CUR?
+CIPAP_CUR:ip:"192.168.4.1"
+CIPAP_CUR:gateway:"192.168.4.1"
+CIPAP_CUR:netmask:"255.255.255.0"

OK
AT+CWSAP_CUR?
AT+CWSAP_CUR?
+CWSAP_CUR:"ESP_A0591D","",1,0,4,0

OK
AT+CIPSERVER=0
AT+CIPSERVER=0
no change

OK
AT+CIPSERVER=1,80
AT+CIPSERVER=1,80

OK
0,CONNECT
1,CONNECT

+IPD,0,439:GET / HTTP/1.1
Host: 192.168.4.1
Connection: keep-alive
Cache-Control: max-age=0
Upgrade-Insecure-Requests: 1
Save-Data: on
User-Agent: Mozilla/5.0 (Windows NT 6.1; WOW64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/57.0.2987.137 YaBrowser/17.4.1.919 Yowser/2.5 Safari/537.36
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,image/webp,*/*;q=0.8
Accept-Encoding: gzip, deflate, sdch
Accept-Language: ru,en;q=0.8

AT+CIPSEND=,308
AT+CIPSEND=,308

ERROR

MaJerle, 2017-06-09:
Hi, problem is this: 'AT+CIPSEND=,308' as there is missing connection number. Now we have to find what is causing this problem. Btw, how do you call Send data function?

lavruhin, 2017-06-09:
I took the function of sending data from your example (01-EXAMPLE_SERVER):

            if (ESP_IsReady(&ESP) == espOK) { 
                if (strstr((char *)data, "/favicon")) {
                    ESP_CONN_Close(&ESP, conn, 0);
                } else {
                    espRes = ESP_CONN_Send(&ESP, conn, responseData, sizeof(responseData), &bw, 0);
                }
            }

lavruhin, 2017-06-09:
Indeed, 
AT + CIPSEND = 0,308 
ends successfully.

I tried to work with the library a year аgо. Then there was a pause in my project. When I returned to this, a new firmware and a new library were released. Now, I see, the concept has changed: ESP_LL_Callback appeared. I still use ESP_LL_SendData.

MaJerle, 2017-06-09:
It is not yet released. You have to clone repo or download it from main code page on repo.

lavruhin, 2017-06-09:
OK, thanks.