MaJerle/ESP8266_AT_Commands_parser #52: Add possibility of manual setting ESP_MAX_CONNECTIONS in config file

d21d3q, 2017-06-30:
Motivation:
Having `ESP_MAX_CONNECTIONS` set to 5 causes `ESP` struct to be more than 7kb while having device in server mode with eg two connections may be sufficient. 

suggested implementation: 
```
/* This settings should not be modified */
#if ESP_SINGLE_CONN
#define ESP_MAX_CONNECTIONS         1   /*!< Number of maximum active connections on ESP */
#else
#ifndef ESP_MAX_CONNECTIONS
#define ESP_MAX_CONNECTIONS         5   /*!< Number of maximum active connections on ESP */
#endif
#endif /* ESP_SINGLE_CONN */
```

MaJerle, 2017-06-30:
The problem is, what if you are server and ESP sends you connection active on number 4. Then you don't have memory and you can only ignore everything. When you are server, you are *slave* for ESP, which is *master* and he controls data. We can only ignore connection. Bad idea.

d21d3q, 2017-06-30:
That makes sense. I didn't check if it is possible to limit  connections on ESP side... 
If we had [AP only](https://github.com/MaJerle/ESP8266_AT_Commands_parser/pull/40) mode working, then we could have single connection? (only one for server)

MaJerle, 2017-06-30:
If you want server mode enabled, you have to use multi connection mode which allows 5 connections from ESP side.

d21d3q, 2017-06-30:
How about closing connection excess connection with some short close message? In such case user is aware of possibility of having connections closed by device... but it sounds like annoying behavior... Do you know if ESP is picking first lowest free connection ID?

MaJerle, 2017-07-14:
ESP is always picking lowest available connection ID for server.