MaJerle/ESP8266_AT_Commands_parser #54: transfer mode (AT+CIPMODE)

RigiKuo, 2017-07-06:
Hello
I use your "01-EXAMPLE_CLIENT_PT" project to test the following server data to test the normal, but the response is slow, about 5 seconds. I hope to be able to respond immediately.
(1) May I ask what is the reason?
(2) transfer mode (AT + CIPMODE) To set to 1: UART-WiFi passthrough mode?
(3) I tested the "01-EXAMPLE_UART_WIFI_RTOS" project again
I do not know how to use this   "ESP_TRANSFER_Send(&ESP, "1234567890\r\n", 12, 1); "

The above questions have trouble you.
Thank you.

/************************************************************************************************************/

GET /weico_api/get_d01.php?device_id=112201 HTTP/1.1\r\n
Host:211.20.172.69\r\n
\r\n

-------------------------------------------------------------------------------------------------------------------
AT+CIPSTATUS
STATUS:2

OK
AT+CIPSTART=0,"TCP","211.20.172.69",80
0,CONNECT

OK
AT+CIPSTATUS
STATUS:3
+CIPSTATUS:0,"TCP","211.20.172.69",80,22988,0

OK
AT+CIPSEND=0,77

OK
> 
Recv 77 bytes

SEND OK

<- - - - - - - - Stop for about 5 seconds

response:

+IPD,0,698,211.20.172.69,80:HTTP/1.1 200 OK
Date: Thu, 06 Jul 2017 07:38:10 GMT
Server: Apache/2.4.25 (Win32) OpenSSL/1.0.2j PHP/5.6.30
X-Powered-By: PHP/5.6.30
Content-Length: 498
Content-Type: text/html; charset=UTF-8

{"success":1,"device_main":[{"id":"1","device_id":"112201","device_name":"ESP_112201","device_password":"1234","device_email":"wee","device_phone":"11112","group_id":"","softap_ssid":"","softap_password":"","softap_ip":"192.168.4.1","softap_mac":"AA:BB:CC:11:22:01","local_ssid":"","local_password":"","local_ip":"192.168.1.100","local_mac":"test","cloud_name":"","cloud_password":"","cloud_ip":"211.20.172.69","cloud_dns":"","created_at":"2017-07-03 17:26:21","updated_at":"2017-07-06 11:57:20"}]}0,CLOSED

MaJerle, 2017-07-06:
There is issue you server side. Data were sent, you can see by receiving `SENT OK` notification. Now why it took 5 seconds to respond, this is not library case. I believe server needed time to do something before respond.

In WIFI PASSTHROUGH mode, everything sent to UART is directly forwarded as packet and sent via network. Works in other direction too, just like normal UART bridge.

RigiKuo, 2017-07-06:
Thank you for your answer.

But I used the following at command test to react immediately.

May I ask what is the reason?

/***********************************************************************************************************/
AT

OK
AT+GMR
AT version:1.4.0.0(May  5 2017 16:10:59)
SDK version:2.1.0(116b762)
compile time:May  5 2017 16:37:51
OK
AT+CWMODE=1

OK
AT+CWJAP="WEICO-WIFI","weico0617a"
WIFI DISCONNECT
WIFI CONNECTED
WIFI GOT IP

OK
AT+CWAUTOCONN=1

OK
AT+CIPSTART="TCP","211.20.172.69",80
CONNECT

OK
AT+CIPMODE=1

OK
AT+CIPSEND

OK

>

-------------------------------------------------------------------------------------------------------------------
GET /weico_api/get_d01.php?device_id=112201 HTTP/1.1\r\n
Host:211.20.172.69\r\n
\r\n
-------------------------------------------------------------------------------------------------------------------

HTTP/1.1 200 OK
Date: Thu, 06 Jul 2017 08:31:27 GMT
Server: Apache/2.4.25 (Win32) OpenSSL/1.0.2j PHP/5.6.30
X-Powered-By: PHP/5.6.30
Content-Length: 498
Content-Type: text/html; charset=UTF-8

{"success":1,"device_main":[{"id":"1","device_id":"112201","device_name":"ESP_112201","device_password":"1234","device_email":"wee","device_phone":"11112","group_id":"","softap_ssid":"","softap_password":"","softap_ip":"192.168.4.1","softap_mac":"AA:BB:CC:11:22:01","local_ssid":"","local_password":"","local_ip":"192.168.1.100","local_mac":"test","cloud_name":"","cloud_password":"","cloud_ip":"211.20.172.69","cloud_dns":"","created_at":"2017-07-03 17:26:21","updated_at":"2017-07-06 11:57:20"}]}

MaJerle, 2017-07-06:
I don't know what is the reason.

RigiKuo, 2017-07-06:
Thank you for answering
How to use the "01-EXAMPLE_UART_WIFI_RTOS" project, "ESP_TRANSFER_Send (& ESP," 1234567890 \ r \ n ", 12, 1); How do I bring the following to ESP_TRANSFER_Send.

------------------------------------------------------------------------------------------------------------------
GET /weico_api/get_d01.php?device_id=112201 HTTP/1.1\r\n
Host:211.20.172.69\r\n
\r\n
------------------------------------------------------------------------------------------------------------------

MaJerle, 2017-07-06:
Like this:

```
//Define variable
const char myData[] = "GET /weico_api/get_d01.php?device_id=112201 HTTP/1.1\r\n"
"Host:211.20.172.69\r\n"
"\r\n";
```

And later in your code use `ESP_TRANSFER_Send(&ESP, myData, strlen(myData), 1);`

RigiKuo, 2017-07-06:
Yes, the problem is solved! Thank you very much for your solution

/ * ------------------------------------------------ -------------------------------------------------- -------- * /
(01) "01-EXAMPLE_UART_WIFI_RTOS" project Response content:

> HTTP / 1.1 200 OK
Date: Thu, 06 Jul 2017 10:44:13 GMT
Server: Apache / 2.4.25 (Win32) OpenSSL / 1.0.2j PHP / 5.6.30
X-Powered-By: PHP / 5.6.30
Content-Length: 498
Content-Type: text / html; charset = UTF-8

(02) "01-EXAMPLE_CLIENT_PT" project Response Content:

+ IPD, 0,698,211.20.172.69,80: HTTP / 1.1 200 OK
Date: Thu, 06 Jul 2017 10:47:05 GMT
Server: Apache / 2.4.25 (Win32) OpenSSL / 1.0.2j PHP / 5.6.30
X-Powered-By: PHP / 5.6.30
Content-Length: 498
Content-Type: text / html; charset = UTF-8

/ * ------------------------------------------------ -------------------------------------------------- -------- * /

Response content less + IPD, 0,698,211.20.172.69,80: immediate response

Is the problem set by AT + CIPMODE transfer mode?

MaJerle, 2017-07-06:
As you can see, your formatting is strange, please use Github text editor to follow nice formatting.

I wouldn't know, this is something we should ask on Espressif forums for AT commands.

closed 2017-07-06 by MaJerle

RigiKuo, 2017-07-06:
Okay!
I tried to "01-EXAMPLE_UART_WIFI_RTOS" project
ESP_TRANSFER_Send (& ESP, myData, strlen (myData), 1); add to my project, test

The following message appears after compiling
. \ Targets \ F411_NUCLEO_HSE \ project.axf: Error: L6218E: Undefined symbol ESP_TRANSFER_Send (referred by main.o).

May I ask what is the reason?

RigiKuo, 2017-07-07:
I found the problem

esp8266_config.h

(1)#define ESP_SINGLE_CONN = 1 
(2)#define ESP_RTOS = 1

Thank you

MaJerle, 2017-07-07:
Yes, you have to be in single connection mode to use transfer feature.

RigiKuo, 2017-07-12:
Hello

I encountered a problem is "ESP8266.c" library under

/* - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - */
/* estatic PT_THREAD (PT_Thread_BASIC (struct pt * pt, evol ESP_t * ESP)) */

   else if (ESP-> ActiveCmd == CMD_BASIC_RST)
   PT_WAIT_UNTIL (pt, ESP-> Time - time> 2)

/* - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - -*/
/* estatic PT_THREAD (PT_Thread_TCPIP (struct pt * pt, evol ESP_t * ESP)) */

   else if (ESP-> ActiveCmd == CMD_TCPIP_TRANSFER_STOP)
   PT_WAIT_UNTIL (pt, (ESP-> Time - btw)> 100);
   
/* - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - -- - - - - - - - - - - - - - */   

The problem is that ESP-> Time has always been equal to 0, but the test of your project is a numerical value, and the action is normal

What is the problem? How to solve?

The above questions have trouble you! Thank you

MaJerle, 2017-07-12:
If you look closely to examples, you will find UpdateTime function in main.c file which is called from 1ms generated tick.

RigiKuo, 2017-07-13:
The problem has been solved
Thank you very much

/*--------------------------------------------------------------------------------------------------------------*/
void TMR3_IRQHandler(void) {
	 
   /* 1ms handler function, called from SysTick interrupt */  
	 /* Update ESP library time for 1 ms */
	 ESP_UpdateTime(&ESP, 1); WPWRCTL=~WPWRCTL;
	 /* Clear Int Flag */			 
	 TIMER_ClearIntFlag(TIMER3); 	
}

void ESP_UPDATETIME_TIMER3(void)  {	
	
	 /* Give a dummy target frequency here. Will over write prescale and compare value with macro */
   TIMER_Open(TIMER3, TMR3_OPERATING_MODE, TMR3_OPERATING_FREQ);	
	 /* Update prescale and compare value to what we need in event counter mode. */
   TIMER_SET_PRESCALE_VALUE(TIMER3, 0);
   TIMER_SET_CMP_VALUE(TIMER3, 12000);
   /* Counter increase on falling edge */
   TIMER_EnableEventCounter(TIMER0, TIMER_COUNTER_FALLING_EDGE);
   /* Start Timer 0 */
   TIMER_Start(TIMER3);
   /* Enable timer interrupt */
   TIMER_EnableInt(TIMER3);
   NVIC_EnableIRQ(TMR3_IRQn);

}

RigiKuo, 2017-07-13:
Hello
i use your "01-EXAMPLE_SERVER" and "01-EXAMPLE_CLIENT_PT" project design, esp8266_config.h single connection mode is defined in #define ESP_SINGLE_CONN disables (0) disabled mode work

I added the ESP_TRANSFER_Start transfer method for the "01-EXAMPLE_UART_WIFI_RTOS" project, but was defined in #define ESP_SINGLE_CONN enabled (1)

My question is whether the project can be free to switch the single-link mode enabled and disabled?

looking forward to receiving your reply soon, in advance

MaJerle, 2017-07-13:
RigiKuo, this issue is closed.

RigiKuo, 2017-07-13:
sorry, i can not understan it.
why the issue is closed

MaJerle, 2017-07-13:
Please take some time and read what is purpose of "issue" on Github. 
You had opening question about your problem and it was fixed. Now issue is closed. 

For new issue open new ticket.

But from your last message about problem, I can't understand exactly what is the problem. If you have 2 ESP devices and one in transfer mode, second in multiple connection mode, this is possible but requires some programming of high level API (user code).

RigiKuo, 2017-07-13:
Okay!

I understand that I will read the study of your project and issues carefully

Thank you