MaJerle/ESP8266_AT_Commands_parser #60: Configuration errors Keil uVision

Ahmedkdk, 2017-11-07:
Hi MaJerle,
I downloaded your library and opened some of the example projects, I sat the clock frequency in the define.h (Not fully sure about that) then I selected my target board (STM32F446RE) but everytime I compile an example I get the same 690 errors, Some of them are below. I have a feeling they all have one thing in common, not sure what. Looking forward to your help if you are familiar with this.
Regards

..\00-HAL_DRIVERS\STM32F4xx_HAL_Driver\Inc\stm32f4xx_hal_rcc_ex.h(320): error:  #256: invalid redeclaration of type name "RCC_PLLI2SInitTypeDef"  (declared at line 115)   }RCC_PLLI2SInitTypeDef;
..\00-HAL_DRIVERS\STM32F4xx_HAL_Driver\Inc\stm32f4xx_hal_rcc_ex.h(337): error:  #256: invalid redeclaration of type name "RCC_PeriphCLKInitTypeDef"  (declared at line 189)
  }RCC_PeriphCLKInitTypeDef;
..\00-HAL_DRIVERS\STM32F4xx_HAL_Driver\Inc\stm32f4xx_hal_rcc_ex.h(408): warning:  #47-D: incompatible redefinition of macro "RCC_PERIPHCLK_RTC"  (declared at line 368)
..\00-HAL_DRIVERS\STM32F4xx_HAL_Driver\Inc\stm32f4xx_ll_fmc.h(765): warning:  #47-D: incompatible redefinition of macro "FMC_NAND_TypeDef"  (declared at line 563 of "..\00-HAL_DRIVERS\STM32F4xx_HAL_Driver\Inc\stm32f4xx_ll_fsmc.h")
     #define FMC_NAND_TypeDef               FMC_Bank3_TypeDef

MaJerle, 2017-11-07:
Sounds like you have multiple definitions for differnet mcus at the same time. Check global compiler macros (in keil go to c/c++ tab in settings) and search if there is any suspicious definition of mcu family which shoupd not be used for f446

Ahmedkdk, 2017-11-07:
Thank you for the quick response.
In the target selection list on the right side of [Download] button STM32F446RE is not stated, so I selected the STM32F4_Discovery board and on the 'Options for Target' dialog I selected the STM32F446RE.
I opened c/c++ tab and I saw STM32F407xx and the Discovery written in the 'Processor Symbol Define' log there so I erased them and wrote STM32F446RE. So I dont think I have done that correctly because when I compiled I got some 16 errors that look like this:  
..\00-STM32_LIBRARIES\stm32fxxx_hal.h(197): error:  #35: #error directive: "There is not selected STM32 family used. Check stm32fxxx_hal.h file for configuration!"

[image]

[image]

MaJerle, 2017-11-07:
Open file "stm32f4xx.h" and on top you will find what define must be set for your MCU.

`  /* #define STM32F446xx */   /*!< STM32F446MC, STM32F446ME, STM32F446RC, STM32F446RE, STM32F446VC, STM32F446VE, STM32F446ZC, and STM32F446ZE Devices */`

Then, I can see you have a define `STM32F4_DISCOVERY`. This define is used for my DISCO lib to know where button is connected and where LEDs are. Since I don't know which board you have (custom, etc), remove this define and also remove TM DISCO library from project and also remove all functions related to this library.

Then it should work.

Ahmedkdk, 2017-11-07:
Sorry, dont mind the right side of the print screen. Only the left side with Keil

Ahmedkdk, 2017-11-07:
So I did accordingly, and managed to set the pins for my device in the by following the Nucleo STM411 that you have given an example on.. But then I got this error:

.\Targets\F411_NUCLEO_HSE\project.axf: error: L6050U: The code size of this image (36520 bytes) exceeds the maximum allowed for this version of the linker.

Is there another way to go around this error other than buying the software?
Regards

MaJerle, 2017-11-07:
Unfortunately not. It is only possible if you are using STM32F0 or STM32L0. In this case, you have full license free of charge.

Ahmedkdk, 2017-11-07:
Can you give any instructions on how to run your examples on IAR Embedded Workbench IDE? I would be grateful.
I used the IAR converter but it isn't working

closed 2017-11-09 by Ahmedkdk