MaJerle/ESP8266_AT_Commands_parser #62: port_cm4.c macros

Ahmedkdk, 2017-11-15:
IDE: IAR Embedded Workbench ARM8.11.3
Device: Nucleo STM32F401RE

This error appears after compilation, What could the problem be? Thanks in advance

Fatal Error[Pe035]: #error directive: This port can only be used when the project options are configured to enable hardware floating point support. E:\IT\Wireless Network Creation\ESP8266_AT_Commands_parser-master (1)\ESP8266_AT_Commands_parser-master\00-STM32_LIBRARIES\FreeRTOS\portable\RVDS\ARM_CM4F\port_cm4.c 79

MaJerle, 2017-11-15:
For IAR you have to use another port file. Please read more on freertos website which port file must be used

Ahmedkdk, 2017-11-16:
I tried before to change the directory to the IAR port directory and the code produced a similar error, I just found out that changing the directory address in the options window is not enough but I also had to remove the already built port.c file under the FreeRTOS group in the workspace window on the left and "add" the new port.c file from the IAR port directory. 
Thanks for the hint.

closed 2017-11-16 by Ahmedkdk