MaJerle/ESP8266_AT_Commands_parser #64: Failure to read stdio.h

Ahmedkdk, 2017-11-17:
Hi, the error is coming from the lines below in the main.c file of almost every example:

int fputc(int ch, FILE *f) {

Even when I added the header stdio.h which contains the FILE structure, the compiler kept producing this error:

Error[Pe020]: identifier "FILE" is undefined E:\IT\Wireless Network Creation\01-Networ_SCAN_From_Keil\User\main.c 168 

Seeking your help with the issue.
Regards
Ahmed

Ahmedkdk, 2017-11-17:
Could it be a circular dependency issue?

MaJerle, 2017-11-17:
This is toolchain related ossue, not this lib issue. Depends on your toolchain, you have different solitions.

Ahmedkdk, 2017-11-17:
I see, so I want to trace the FILE pointer references to trace how the compiler is calling the stdio.h library all the way from the main.c file. Since I have the following headers in the main file:
/* Include core modules */
#include "stm32fxxx_hal.h"
#include "defines.h"
#include "tm_stm32_disco.h"
#include "tm_stm32_delay.h"
#include "tm_stm32_usart.h"
#include "esp8266.h"
#include "cmsis_os.h"

I think the stdio.h is called through main>esp8266>stdio.h 
that seems normal, and it shouldnt produce any error. Can you give me any suggestion of how to approach this?

Regards