MaJerle/ESP8266_AT_Commands_parser #66: ESP support for STM32F429ZI-Nucleo

paiarunk, 2017-12-05:
I am trying to use the same for STM32F429ZI-Nucleo board. 

Please help

Arun

MaJerle, 2017-12-05:
I dont see how I can help.

paiarunk, 2017-12-05:
Hi

I am trying to port the sample for STM32F429ZI-Nucleo board, but not
able to test a GPIO toggle also. So any suggestions would be helpful..

Arun

MaJerle, 2017-12-05:
You have ti specify what tests you did, how you started. You nesd to help me if you want my help otherwise I dont know where to start. Maybe you shouod start by leds, then moving to something more?

paiarunk, 2018-01-01:
I have successfully tested the esp8266 in Keil with STM32F429ZI-Nucleo board, but I am trying to use Coocox CoIDE for our further development, But in CoIDE I am finding PT_THREAD not running and ESP8266 not initalizing. 

Any suggestions will be helpful

Arun

MaJerle, 2018-01-02:
PT_THREAD is pure C language code and is perfectly valid to use. Now the question is how you did tests.
Support on this library does not exists anymore from my side, because I moved to new new for OS systems only.

https://github.com/MaJerle/ESP_AT_Lib

paiarunk, 2018-01-02:
Hi

Thanks for the mail.

Can I share the project so that you can help me?

Basically I have listed all the files from keil ide project and added with
changes in start-up assembly file required for gcc compiler.

All my gpio, spi and sd card etc is working fine as in keil, but only the
esp thread is not running.

Also I found the  new library is compatible in truestudio and keil,
I will try the same.

Please help me. With your suggestions

paiarunk, 2018-01-03:
Hi TM

I need some help on how to pass the values in the function
TM_RCC_InitSystem in CoIDE

Since the values in Keil are sent as Preprocessor values, I think this
is required .

,HSE_VALUE=8000000,RCC_OSCILLATORTYPE=RCC_OSCILLATORTYPE_HSE,RCC_PLLSOURCE=RCC_PLLSOURCE_HSE,RCC_PLLM=8,RCC_PLLN=360,RCC_PLLP=2,RCC_PLLQ=7,

Please help.

Regards

R Arun Kumar

MaJerle, 2018-01-03:
You need to add macros to your toolchain or edit `defines.h` file.

http://www.coocox.org/book/coocox/coide-dev-manual/4-Project-Management/4.4-Project-configuration

I don't want to go to the details here as:
1. Problem is not relevant with AT library
2. This library is not supported anymore from my side because new one is in development and can be used already.

closed 2018-01-03 by MaJerle